stereolabs/zed-opencv #2: Installing_OpenCV on Jetson TK1 error

won13y, 2016-01-27:
I tried to install opencv on Jetson TK1 
as described in http://elinux.org/Jetson/Installing_OpenCV

When I type sudo make -j4 install, error is as followings:

ubuntu@tegra-ubuntu:~/opencv-2.4.9/build$ sudo make -j4 install

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(51): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(52): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(53): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(54): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(55): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(56): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(57): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(58): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(61): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(62): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(63): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(64): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(65): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(66): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(67): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(68): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(119): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(120): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(121): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(122): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(123): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(124): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(125): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(126): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(127): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(128): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(129): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(130): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(131): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(132): error: a     storage class is not allowed in an explicit specialization

    /home/ubuntu/opencv-2.4.9/modules/gpu/src/nvidia/core/NCVPixelOperations.hpp(133): error: a     storage class is not allowed in an explicit specialization

    31 errors detected in the compilation of "/tmp/tmpxft_00007bdf_00000000-6_NCVPyramid.cpp1.ii".
    CMake Error at cuda_compile_generated_NCVPyramid.cu.o.cmake:264 (message):
      Error generating file
      /home/ubuntu/opencv-2.4.9/build/modules/gpu/CMakeFiles/cuda_compile.dir/src/nvidia/core/.    /cuda_compile_generated_NCVPyramid.cu.o

    make[2]: *** [modules/gpu/CMakeFiles/cuda_compile.dir/src/nvidia/core/.    /cuda_compile_generated_NCVPyramid.cu.o] Error 1
    make[2]: *** Waiting for unfinished jobs....
    make[1]: *** [modules/gpu/CMakeFiles/opencv_gpu.dir/all] Error 2
    make: *** [all] Error 2

won13y, 2016-01-27:
I cannot compile OpenCV2.4.9 due to the above circumstance 

but, OpenCV2.4.10.1 is compiled correctly on Jetson TK1,

Is  OpenCV2.4.9 only compatible with ZED camera?

In ZED_Developer_Guide.pdf 

it only refer opencv 2.4.9

    Installing the ZED SDK – Linux Before you start developing applications witht he ZED, download     and install CUDA 6.5 toolkit .Compile and install OpenCV2.4.9

    https://www.stereolabs.com/developers/documentation/ZED_Developer_Guide.pdf

won13y, 2016-01-27:
I fount solution but it seems incomplete

I change -DWITH_CUDA=ON  -> OFF

as described in http://elinux.org/Jetson/Installing_OpenCV

But I don't know whether it is complete solution or not.

    cmake -DWITH_CUDA=ON -DCUDA_ARCH_BIN="3.2" -DCUDA_ARCH_PTX=""     -DBUILD_TESTS=OFF -DBUILD_PERF_TESTS=OFF ..

    cmake -DWITH_CUDA=OFF -DCUDA_ARCH_BIN="3.2" -DCUDA_ARCH_PTX=""     -DBUILD_TESTS=OFF -DBUILD_PERF_TESTS=OFF ..

axjensen, 2016-01-27:
I use OpenCV  version 2.4.11 on Jetson TK1 and everything seems to work fine.  I believe, that I use these settings for cmake (I can check later):
cmake -DWITH_CUDA=ON -DCUDA_ARCH_BIN="3.2" -DCUDA_ARCH_PTX="3.0" -DBUILD_TESTS=OFF -DBUILD_PERF_TESTS=OFF

adujardin, 2016-01-27:
Hi,

The ZED SDK is compatible with OpenCV 2.4.X (>=2.4.9 tested). On the Jetson, we only use the version provided by the Jetpack (closed source but optimized by Nvidia).

However, the solution provided by @axjensen should work. By default the jetson gpu arch (3.2) is not built.

You can also disable CUDA support for OpenCV if that solve your compilation issues and if you don't use it. The ZED SDK doesn't need it.

won13y, 2016-01-29:
@adujardin  Did you mean that ZED SDK can be used with OpenCV 2.4.X (>=2.4.9 tested)?
I will try openCV 2.4.11 and let you know the results. Thank you

closed 2016-02-15 by adujardin