stereolabs/zed-opencv #7: Problem only in Desktop folder(ZED SDK >> Resolution not yet available)

won13y, 2016-03-18:
I have successfully run zed-opencv example in /usr/local/zed/sample/Y/cpu folder

but when I copy the same folder(zed-opencv) in Desktop folder,

it always results in following error:

ZED SDK >> Resolution not yet available

, which means that ZED is not available.

But, zed-opencv example in /usr/local/zed/sample/Y/cpu folder always work correctly.

I don't know why.

It seems to me that this problem might be related to authority problem?

adujardin, 2016-03-18:
Hi, 

Normally, this error appears on the Jetson when the resolution is set higher than 720p (excluded). Could you check the resolution in the main file?

won13y, 2016-03-19:
It is the same source file that work correctly in  /usr/local/zed/sample/Y/cpu folder, 

but the very same source file copied in Desktop folder produces the following errors:

ZED SDK >> Resolution not yet available

In the main file,

the resolution is set to be VGA.

closed 2022-04-27 by github-actions[bot]