stereolabs/zed-opencv #8: Using OpenCV 3.1

eduardohenriquearnold, 2016-05-17:
Hi. I am using the latest SDK Version, 0.94. Although the ZED framework uses OpenCV 2.4, which I have compiled, my application was developed using OpenCV 3.1 and I cannot afford to adapt it to the previous version.

I am able to run the ZED apps such as _Depth Viewer_ and _Explorer_ with the 2.4 so files I've placed in the ZED libs folder. When I try to compile an application, however, using the sample CMake script it simply says that `The following configuration files were considered but not accepted:  /usr/local/share/OpenCV/OpenCVConfig.cmake, version: 3.1.0`.

Another try was to try to link both OpenCV 2.4 and 3.1 libs and the result was 
```
g++ -o record record.o `pkg-config --libs opencv zed cuda-7.5`
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by /usr/local/zed/lib/libsl_zed.so, may conflict with libopencv_core.so.3.1
/usr/bin/ld: warning: libopencv_highgui.so.2.4, needed by /usr/local/zed/lib/libsl_zed.so, may conflict with libopencv_highgui.so.3.1
/usr/bin/ld: warning: libopencv_features2d.so.2.4, needed by /usr/local/zed/lib/libopencv_calib3d.so.2.4, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libopencv_flann.so.2.4, needed by /usr/local/zed/lib/libopencv_calib3d.so.2.4, not found (try using -rpath or -rpath-link)
/usr/local/zed/lib/libopencv_calib3d.so.2.4: undefined reference to `cv::flann::Index::~Index()'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cv::imwrite(std::string const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
/usr/local/zed/lib/libopencv_calib3d.so.2.4: undefined reference to `cv::SimpleBlobDetector::SimpleBlobDetector(cv::SimpleBlobDetector::Params const&)'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cv::imshow(std::string const&, cv::_InputArray const&)'
/usr/local/zed/lib/libopencv_calib3d.so.2.4: undefined reference to `cv::SimpleBlobDetector::Params::Params()'
/usr/local/zed/lib/libopencv_calib3d.so.2.4: undefined reference to `cv::flann::LinearIndexParams::LinearIndexParams()'
/usr/local/zed/lib/libopencv_calib3d.so.2.4: undefined reference to `cv::flann::Index::Index(cv::_InputArray const&, cv::flann::IndexParams const&, cvflann::flann_distance_t)'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cv::namedWindow(std::string const&, int)'
/usr/local/zed/lib/libopencv_calib3d.so.2.4: undefined reference to `cv::flann::IndexParams::~IndexParams()'
/usr/local/zed/lib/libopencv_calib3d.so.2.4: undefined reference to `cv::flann::Index::knnSearch(cv::_InputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, int, cv::flann::SearchParams const&)'
/usr/local/zed/lib/libsl_calibration.so: undefined reference to `cv::Exception::Exception(int, std::string const&, std::string const&, std::string const&, int)'
/usr/local/zed/lib/libopencv_calib3d.so.2.4: undefined reference to `cv::FeatureDetector::detect(cv::Mat const&, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >&, cv::Mat const&) const'
/usr/local/zed/lib/libopencv_calib3d.so.2.4: undefined reference to `cv::flann::SearchParams::SearchParams(int, float, bool)'
collect2: error: ld returned 1 exit status
Makefile:13: recipe for target 'record' failed
make: *** [record] Error 1
```
PS: I've written a .pc file to make it easier to use Make, the contents can be seen [here](https://gist.github.com/eduardohenriquearnold/102f771cdc17d8e8010f1e34cf1ca544).

eduardohenriquearnold, 2016-05-17:
I think I have almost managed to make it work. By using OpenCV 2.4.12 instead of 2.4.9 as the documentation suggested and placing __core, calib3d, highgui, imgproc__ libs inside ZED lib folder, I was able to reduce the errors to:
```
g++ -o record record.o `pkg-config --libs opencv zed cuda-7.5`
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by /usr/local/zed/lib/libsl_zed.so, may conflict with libopencv_core.so.3.1
/usr/bin/ld: warning: libopencv_highgui.so.2.4, needed by /usr/local/zed/lib/libsl_zed.so, may conflict with libopencv_highgui.so.3.1
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cv::imwrite(std::string const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cv::imshow(std::string const&, cv::_InputArray const&)'
/usr/local/zed/lib/libsl_zed.so: undefined reference to `cv::namedWindow(std::string const&, int)'
/usr/local/zed/lib/libsl_calibration.so: undefined reference to `cv::Exception::Exception(int, std::string const&, std::string const&, std::string const&, int)'
collect2: error: ld returned 1 exit status
Makefile:13: recipe for target 'record' failed
make: *** [record] Error 1
```
I also tried the following:
```
/usr/local/zed/lib$ ldd -r libsl_zed.so 
	linux-vdso.so.1 =>  (0x00007ffd9c9a3000)
	libcudart.so.7.5 => /usr/local/cuda/lib64/libcudart.so.7.5 (0x00007fd2830f8000)
	libsl_calibration.so => /usr/local/zed/lib/libsl_calibration.so (0x00007fd282e05000)
	libsl_depthcore.so => /usr/local/zed/lib/libsl_depthcore.so (0x00007fd28289e000)
	libopencv_core.so.2.4 => /usr/local/zed/lib/libopencv_core.so.2.4 (0x00007fd2822ac000)
	libopencv_highgui.so.2.4 => /usr/local/zed/lib/libopencv_highgui.so.2.4 (0x00007fd281f55000)
	libopencv_imgproc.so.2.4 => /usr/local/zed/lib/libopencv_imgproc.so.2.4 (0x00007fd281a45000)
	libopencv_calib3d.so.2.4 => /usr/local/zed/lib/libopencv_calib3d.so.2.4 (0x00007fd28176d000)
	libnppc.so.7.5 => /usr/local/cuda/lib64/libnppc.so.7.5 (0x00007fd281504000)
	libnppi.so.7.5 => /usr/local/cuda/lib64/libnppi.so.7.5 (0x00007fd27d632000)
	libnpps.so.7.5 => /usr/local/cuda/lib64/libnpps.so.7.5 (0x00007fd27ce45000)
	libusb-0.1.so.4 => /lib/x86_64-linux-gnu/libusb-0.1.so.4 (0x00007fd27cc20000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd27c89d000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd27c595000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd27c37e000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd27bfb3000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fd27bdaf000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd27bb91000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fd27b988000)
	libopencv_video.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4 (0x00007fd27b732000)
	libcudpp.so => /usr/local/zed/lib/libcudpp.so (0x00007fd271e0a000)
	libcuda.so.1 => /usr/lib/x86_64-linux-gnu/libcuda.so.1 (0x00007fd270e55000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fd270c3a000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007fd2709e5000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007fd2707be000)
	libtiff.so.5 => /usr/lib/x86_64-linux-gnu/libtiff.so.5 (0x00007fd27054b000)
	libjasper.so.1 => /usr/lib/x86_64-linux-gnu/libjasper.so.1 (0x00007fd2702f4000)
	libgtk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0 (0x00007fd26fca7000)
	libgdk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0 (0x00007fd26f9f2000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007fd26f79f000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fd26f48f000)
	libdc1394.so.22 => /usr/lib/x86_64-linux-gnu/libdc1394.so.22 (0x00007fd26f21b000)
	libavcodec-ffmpeg.so.56 => /usr/lib/x86_64-linux-gnu/libavcodec-ffmpeg.so.56 (0x00007fd26de76000)
	libavformat-ffmpeg.so.56 => /usr/lib/x86_64-linux-gnu/libavformat-ffmpeg.so.56 (0x00007fd26da89000)
	libavutil-ffmpeg.so.54 => /usr/lib/x86_64-linux-gnu/libavutil-ffmpeg.so.54 (0x00007fd26d81b000)
	libswscale-ffmpeg.so.3 => /usr/lib/x86_64-linux-gnu/libswscale-ffmpeg.so.3 (0x00007fd26d59b000)
	libopencv_features2d.so.2.4 => /home/eduardo/Documents/ZED/opencv-2.4.12/build/lib/libopencv_features2d.so.2.4 (0x00007fd26d2ef000)
	libopencv_flann.so.2.4 => /home/eduardo/Documents/ZED/opencv-2.4.12/build/lib/libopencv_flann.so.2.4 (0x00007fd26d07b000)
	/lib64/ld-linux-x86-64.so.2 (0x0000559353322000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007fd26ce58000)
	libjbig.so.0 => /usr/lib/x86_64-linux-gnu/libjbig.so.0 (0x00007fd26cc4a000)
	libgmodule-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007fd26ca45000)
	libpangocairo-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so.0 (0x00007fd26c838000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007fd26c4fe000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007fd26c2f7000)
	libatk-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libatk-1.0.so.0 (0x00007fd26c0d2000)
	libcairo.so.2 => /usr/lib/x86_64-linux-gnu/libcairo.so.2 (0x00007fd26bdbf000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0 (0x00007fd26bb9c000)
	libgio-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0 (0x00007fd26b81b000)
	libpangoft2-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0 (0x00007fd26b605000)
[... cut ...]
	libicudata.so.55 => /usr/lib/x86_64-linux-gnu/libicudata.so.55 (0x00007fd25e561000)
undefined symbol: _ZN2cv9ExceptionC1EiRKSsS2_S2_i	(/usr/local/zed/lib/libsl_calibration.so)
undefined symbol: _ZN2cv7imwriteERKSsRKNS_11_InputArrayERKSt6vectorIiSaIiEE	(./libsl_zed.so)
undefined symbol: _ZN2cv6imshowERKSsRKNS_11_InputArrayE	(./libsl_zed.so)
undefined symbol: _ZN2cv11namedWindowERKSsi	(./libsl_zed.so)
```
If you check, the __imwrite, namedWindow__ methods were also undefined on the previous output. Are you sure the OpenCV version you are using is 2.4.9?

eduardohenriquearnold, 2016-05-17:
I am trying to find the cause of these undefined symbols. As you can see from the output below, the __imshow__ method is undefined in libsl_zed.so, and it is present in libopencv_highgui.so. However, the prototype seems different, which I think is the cause of these issues. Maybe it is related to the use of C++11? Anyway, I have removed the C++11 flag when compiling and still get the same errors.

```
eduardo@PC:/usr/local/zed/lib$ nm --demangle libopencv_highgui.so.2.4 | grep imshow
00000000000578c0 T cv::imshow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::_InputArray const&)
eduardo@PC:/usr/local/zed/lib$ nm --demangle libsl_zed.so | grep imshow                 
U cv::imshow(std::string const&, cv::_InputArray const&)
```

eduardohenriquearnold, 2016-05-18:
Turns out the version of OpenCV I was using in the previous posts was 2.4.12. I have now compiled 2.4.9, which seems to have fixed that prototype error seen in https://github.com/stereolabs/zed-opencv/issues/8#issuecomment-219815408. 

I have placed the core, imgproc, calib3d and video libs inside ZED libs folder. When I use ldd -r libsl_zed.so, all shared libraries and symbols are found. However, when I try the same with libsl_calibration.so, it does not find any of the opencv libraries, as seen here:
```
eduardo@PC:/usr/local/zed/lib$ ldd libsl_zed.so | grep opencv
	libopencv_core.so.2.4 => /usr/local/zed/lib/libopencv_core.so.2.4 (0x00007f5917fa5000)
	libopencv_highgui.so.2.4 => /usr/local/zed/lib/libopencv_highgui.so.2.4 (0x00007f5917c4c000)
	libopencv_imgproc.so.2.4 => /usr/local/zed/lib/libopencv_imgproc.so.2.4 (0x00007f591773b000)
	libopencv_calib3d.so.2.4 => /usr/local/zed/lib/libopencv_calib3d.so.2.4 (0x00007f591749d000)
	libopencv_video.so.2.4 => /usr/local/zed/lib/libopencv_video.so.2.4 (0x00007f591145b000)
	libopencv_features2d.so.2.4 => /home/eduardo/Documents/ZED/opencv-2.4.9/build/lib/libopencv_features2d.so.2.4 (0x00007f5903016000)
	libopencv_flann.so.2.4 => /home/eduardo/Documents/ZED/opencv-2.4.9/build/lib/libopencv_flann.so.2.4 (0x00007f5902d9c000)
eduardo@PC:/usr/local/zed/lib$ ldd libsl_calibration.so | grep opencv
	libopencv_video.so.2.4 => not found
	libopencv_imgproc.so.2.4 => not found
	libopencv_core.so.2.4 => not found
	libopencv_calib3d.so.2.4 => not found
```
What could be causing this issue? Both files are in the same folder as all the shared libs.

eduardohenriquearnold, 2016-05-19:
I was finally able to compile my application with both OpenCV library versions 3.1 and 2.4.9. Check the library dependencies here:
```
eduardo@PC:~/Documents/RecordRGBD$ ldd record | grep open
	libopencv_highgui.so.3.1 => /usr/local/lib/libopencv_highgui.so.3.1 (0x00007fdae2d57000)
	libopencv_core.so.3.1 => /usr/local/lib/libopencv_core.so.3.1 (0x00007fdae2024000)
	libopencv_imgcodecs.so.3.1 => /usr/local/lib/libopencv_imgcodecs.so.3.1 (0x00007fdadec6f000)
	libopencv_imgproc.so.3.1 => /usr/local/lib/libopencv_imgproc.so.3.1 (0x00007fdadd221000)
	libopencv_core.so.2.4 => /usr/local/zed/lib/libopencv_core.so.2.4 (0x00007fdadb685000)
	libopencv_highgui.so.2.4 => /usr/local/zed/lib/libopencv_highgui.so.2.4 (0x00007fdadb32c000)
	libopencv_imgproc.so.2.4 => /usr/local/zed/lib/libopencv_imgproc.so.2.4 (0x00007fdadae1c000)
	libopencv_calib3d.so.2.4 => /usr/local/zed/lib/libopencv_calib3d.so.2.4 (0x00007fdadab7d000)
	libopencv_video.so.2.4 => /usr/local/zed/lib/libopencv_video.so.2.4 (0x00007fdad1ec6000)
	libopencv_features2d.so.2.4 => /home/eduardo/Documents/ZED/opencv-2.4.9/build/lib/libopencv_features2d.so.2.4 (0x00007fdac63fc000)
	libopencv_flann.so.2.4 => /home/eduardo/Documents/ZED/opencv-2.4.9/build/lib/libopencv_flann.so.2.4 (0x00007fdac6182000)
	libopenjpeg.so.5 => /usr/lib/x86_64-linux-gnu/libopenjpeg.so.5 (0x00007fdac18c0000)
```
However, when running the code I get a Segmentation fault:
```
eduardo@IST-HP-PC:~/Documents/RecordRGBD$ ./record 5000
ZED SDK >> (Init) Best GPU Found : NVS 510 , ID : 0
ZED SDK >> (Init) Disparity mode has been set to Performance mode
ZED SDK >> (Init) Creating ZED GPU mem...
Segmentation fault (core dumped)
```
I have tried using only OpenCV 2.4.9 to build it and all works fine. So the segfault probably happens due to symbol conflict between versions of OpenCV. Following [this guide](http://stackoverflow.com/questions/4539442/can-two-different-versions-of-the-same-libs-with-same-name-exists-in-an-applic), there seems to be a solution to multiple versions of the same library linking. It requires that a StereoLabs developer, @adujardin, rebuild the sl_zed with specific linker flags (`soname` and `default-symver`).

Since many users need newer versions of OpenCV for their applications, would you be kind to check this issue and provide a solution?

Regards

mpperez, 2016-06-07:
Hi,
I have the same problem with OpenCV 2.4.9/ 2.4.13 and zed 0.94  with zed-opencv-master:

`make -j8
[ 50%] Linking CXX executable "ZED with OpenCV"
CMakeFiles/ZED_with_OpenCV.dir/src/main.o: En la función `main':
/zed-opencv-master/src/main.cpp:110: referencia a `sl::zed::Camera::Camera(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)' sin definir
/usr/local/zed/lib/libsl_zed.so: referencia a `cv::imwrite(std::string const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)' sin definir
/usr/local/zed/lib/libsl_zed.so: referencia a `cv::imshow(std::string const&, cv::_InputArray const&)' sin definir
/usr/local/zed/lib/libsl_zed.so: referencia a `cv::namedWindow(std::string const&, int)' sin definir
/usr/local/zed/lib/libsl_calibration.so: referencia a `cv::Exception::Exception(int, std::string const&, std::string const&, std::string const&, int)' sin definir
collect2: error: ld returned 1 exit status
CMakeFiles/ZED_with_OpenCV.dir/build.make:106: fallo en las instrucciones para el objetivo 'ZED with OpenCV'
make[2]: *** [ZED with OpenCV] Error 1
CMakeFiles/Makefile2:67: fallo en las instrucciones para el objetivo 'CMakeFiles/ZED_with_OpenCV.dir/all'
make[1]: *** [CMakeFiles/ZED_with_OpenCV.dir/all] Error 2
Makefile:83: fallo en las instrucciones para el objetivo 'all'
make: *** [all] Error 2`

All the applications from /usr/local/zed/tools work perfect but I can't compile anything. Would add support to OpenCV3.xx soon?

Regards

eduardohenriquearnold, 2016-06-07:
These errors seemed to be cause by using a different compiler version to compile OpenCV and the ZED application. In my case I used g++-4.9 to compile OpenCV since the 5.0 is not compatible with CUDA 7.5. Anyway, try compiling your application with the same version of G++ you used to compile OpenCV.

It would be great to be able to use ZED with OpenCV 3.x. Hope the devs are working on it.

adujardin, 2016-06-21:
Hi,
The last version of the ZED SDK should solve the issue since it's now using OpenCV 3.1.

closed 2016-07-21 by adujardin

udayakumar97, 2016-12-21:
Hi,
I am still getting the error "Segementation Fault" when trying to use opencv 3.1 with jetson tx1..Is there a solution for it?

sl-braun, 2016-12-21:
On TX1, please use the OpenCV Version provided by NVIDIA : OpenCV4Tegra. It is based on 2.4.13
Best,
OB

YousefYaghoobi, 2017-03-08:
Hi sl-braun,

I need to use OpenCV-3.1.0 because of tracking module which is only available on 3.x version on Jetson TX1. 
Any help is highly appreciated.
Best regards,

mepix, 2017-07-06:
Hi,

Is there an update or a workaround for this issue?  I, too, am using OpenCV-3.1.0 on the Jetson TX1 for modules not included in the OpenCV4Tegra package.

Thanks,

GPrathap, 2017-07-08:
Hi ,
I am having the same problem here... instated of OpenCV4Tegra is there any workaround for OpenCV 3.1.x for TX2