stereolabs/zed-opencv #9: ZED Camera: Setting Resolution via Python 2.7 and OpenCV

atomoclast, 2016-05-25:
Hello,

I was wondering if there was a way to directly adjust the ZED camera's resolution? I know we can do it through Cheese, so it should be possible to do it with other code vs only using the ZED SDK.

I have tried code such as:

`cap = cv2.VideoCapture(0)

cap.set(3, 4416)
cap.set(4, 1242)`

as well as code such as: 

`cap = cv2.VideoCapture(0)

cap.set(3, 2208)
cap.set(4, 1242)`

The default resolution is still just 640x480 for each frame (or 1280x480 combined).

Has any one had to do this or had any success in changing the resolution via other means by using the ZED as just a USB camera?

Thanks in advance,

Andrew

Bootstrap360, 2016-10-05:
Have you found any solutions?

Bardo91, 2016-10-18:
I am using it in the c++ code and I am able to configure its resolution using:

         zedCamera.set(CV_CAP_PROP_FRAME_WIDTH, width);
         zedCamera.set(CV_CAP_PROP_FRAME_HEIGHT, height);

I assume that it should be also possible from the python code,

Bootstrap360, 2016-10-19:
I got it working in Python.

I forgot that the ZED camera is twice as wide as the typical 16:9 aspect ratio.

```
cap = cv2.VideoCapture(0)

cap.set(3, 2560)
cap.set(4, 720)
```

Bardo91, 2016-10-19:
So... is it closed the issue?

closed 2022-04-27 by github-actions[bot]