stereolabs/zed-opencv #10: Alternating between grabbing frames and recording

eduardohenriquearnold, 2016-06-01:
I am writing an application which requires me to detect movement, thus grabbing frames, and if enough pixels change to start recording an *svo* file.

Following the documentation I saw that to call the *initRecording* method one must not have used *init* before. So, what I am currently doing is to create *zed::Camera*, and initiating it the usual way, grab frames. After I have detected movement I delete this instance and create a new one. However, my created file has size 0, no matter what. In addition to that, if instead of using *init* I use *initRecording* from the first time (remove the two lines below), it works as expected.

Here's a script that can clarify the issue.

```c++
sl::zed::Camera* zed;
sl::zed::SENSING_MODE dm_type = sl::zed::RAW;

//Omit these three lines and it will work.
zed = new sl::zed::Camera(sl::zed::ZEDResolution_mode::VGA);
sl::zed::ERRCODE err = zed->init(sl::zed::MODE::PERFORMANCE, -1, true, false, true);
delete zed;

zed = new sl::zed::Camera(sl::zed::ZEDResolution_mode::VGA);
zed->initRecording(getDateTimeStr()+".svo");

while (true)
{
  zed->record();
  if (elapsedTimeSinceRef() >= 60*recordingDuration)
  {
     zed->stopRecording();
     break;
  }
}
```

It seems that deleting the object is not detaching the camera properly. Can you please give me some advice?

adujardin, 2016-06-02:
Hi,

Indeed, I am able to reproduce the bug.
For now, the 'solution' I tested (on Linux) was to create another pointer for the recording Camera. 

Like this :
```C++
sl::zed::Camera* zed = new sl::zed::Camera(sl::zed::ZEDResolution_mode::VGA);
sl::zed::ERRCODE err = zed->init(sl::zed::MODE::PERFORMANCE, -1, true, false, true);
delete zed;

sl::zed::Camera* zed2 = new sl::zed::Camera(sl::zed::ZEDResolution_mode::VGA);
zed2->initRecording(getDateTimeStr()+".svo");
```

Thanks for letting us know

eduardohenriquearnold, 2016-06-02:
I have tried many combinations of code, the one I am currently working is this:
```c++
#include <stdio.h>
#include <cstdio>

#include <iostream>
#include <ctime>

#include <opencv2/opencv.hpp>
#include <zed/Camera.hpp>

using namespace std;
using namespace cv;

string getDateTimeStr()
{
	time_t t = time(NULL);
	char res[100];
	strftime(res, sizeof(res), "%Y-%m-%d--%H-%M", localtime(&t));
	return string(res);
}

int main(int argc, char* argv[])
{
	//Constants
	const double startRecordingThreshold = 0.1;
	const double recordingDuration = 1;           //Minutes
	const double recordingMinDelay = 0.1;			  	//Minutes

	//Start camera
	time_t tref = time(NULL);
	sl::zed::SENSING_MODE dm_type = sl::zed::RAW;

	auto elapsedTimeSinceRef = [&]()
	{
		return difftime(time(NULL), tref);
	};

	Mat bgnd;

	while (true)
	{
		sl::zed::Camera *zedgrab = new sl::zed::Camera(sl::zed::ZEDResolution_mode::VGA);
		zedgrab->init(sl::zed::MODE::PERFORMANCE, -1, true, false, true);
		tref = time(NULL);

		double m = 0;
		while (m < startRecordingThreshold || elapsedTimeSinceRef() < 60*recordingMinDelay)
		{
			//Get frame
			if (zedgrab->grab(dm_type))
				continue;

			//Retrieve color and depth frames
			Mat gray;
			sl::zed::slMat2cvMat(zedgrab->retrieveImage(sl::zed::SIDE::LEFT_GREY)).convertTo(gray, CV_32FC1, 1./255);

			//Blurs
			blur(gray,gray, Size(5,5));

			//Set bgnd if empty
			if (bgnd.empty())
				gray.copyTo(bgnd);

			//Generate background model and diff frame
			Mat diff;
			accumulateWeighted(gray, bgnd, 0.02);
			absdiff(gray, bgnd, diff);

			//Change coefficient
			m = mean(diff)[0];
			cout << m << endl;
		}

		//Starts recording
		delete zedgrab;
		sl::zed::Camera *zedrec = new sl::zed::Camera(sl::zed::ZEDResolution_mode::VGA);
		zedrec->initRecording(getDateTimeStr()+".svo");
		tref = time(NULL);

		cout << "started recording" << endl;
		while (elapsedTimeSinceRef() < 60*recordingDuration)
		{
			zedrec->record();
			cout << "recording frames " << elapsedTimeSinceRef() << endl;
		}
		cout << "finished recording" << endl;
		zedrec->stopRecording();

		delete zedrec;
	}

	return 0;
}
```

There seems to be some problems with the *record* method. It sometimes segfaults, other times the *elapsedTime* doesn't get updated. I cannot see any obvious problems with the code. Can you try it?

adujardin, 2016-06-03:
I have the same problem as you describe. Since I'm not used to `time` and `difftime`, I used `std::chrono`. Also, I think it's better to put a sleep between two `record()` call, at the frame rate (VGA : 10ms).
This is the code I used : 

```C++
#include <stdio.h>
#include <cstdio>

#include <iostream>
#include <ctime>
#include <chrono>
#include <thread>

#include <opencv2/opencv.hpp>
#include <zed/Camera.hpp>

using namespace std;
using namespace cv;

string getDateTimeStr()
{
    time_t t = time(NULL);
    char res[100];
    strftime(res, sizeof(res), "%Y-%m-%d--%H-%M", localtime(&t));
    return string(res);
}

int main(int argc, char* argv[])
{
    //Constants
    const double startRecordingThreshold = 0.1;
    const double recordingDuration = 1;           //Minutes
    const double recordingMinDelay = 0.1;               //Minutes

    //Start camera
    sl::zed::SENSING_MODE dm_type = sl::zed::RAW;

    auto start = std::chrono::high_resolution_clock::now();
    auto elapsedTimeSinceRef = [&]() {
        return std::chrono::duration_cast<std::chrono::milliseconds>(std::chrono::high_resolution_clock::now() - start).count()*0.001;
    };

    Mat bgnd;

    while (true)    {
        sl::zed::Camera *zedgrab = new sl::zed::Camera(sl::zed::ZEDResolution_mode::VGA);
        zedgrab->init(sl::zed::MODE::PERFORMANCE, -1, true, false, true);
        start = std::chrono::high_resolution_clock::now();

        double m = 0;
        while (m < startRecordingThreshold || elapsedTimeSinceRef() < 60*recordingMinDelay)        {
            //Get frame
            if (zedgrab->grab(dm_type))
                continue;

            //Retrieve color and depth frames
            Mat gray;
            sl::zed::slMat2cvMat(zedgrab->retrieveImage(sl::zed::SIDE::LEFT_GREY)).convertTo(gray, CV_32FC1, 1./255);

            //Blurs
            blur(gray,gray, Size(5,5));

            //Set bgnd if empty
            if (bgnd.empty())
                gray.copyTo(bgnd);

            //Generate background model and diff frame
            Mat diff;
            accumulateWeighted(gray, bgnd, 0.02);
            absdiff(gray, bgnd, diff);

            //Change coefficient
            m = mean(diff)[0];
            cout << m << endl;
        }

        //Starts recording
        delete zedgrab;
        sl::zed::Camera *zedrec = new sl::zed::Camera(sl::zed::ZEDResolution_mode::VGA);
        zedrec->initRecording(getDateTimeStr()+".svo");
        start = std::chrono::high_resolution_clock::now();

        cout << "started recording" << endl;
        while (elapsedTimeSinceRef() < 60*recordingDuration)  {
            zedrec->record();
            cout << "recording frames " << elapsedTimeSinceRef() << endl;
            std::this_thread::sleep_for(std::chrono::milliseconds(10));
        }
        cout << "finished recording" << endl;
        zedrec->stopRecording();

        delete zedrec;
    }

    return 0;
}
```

Even if it's better, I still have some weird behavior sometimes. There's definitely a bug in the destructor and/or in record().

eduardohenriquearnold, 2016-06-06:
Thanks for testing the code, same happens here. How great would it be if we had ZED source code to try to understand what is happening and possibly solve the issue. Currently we can only wait for  StereoLabs @admin-stereolabs devs to hopefully solve the issue ASAP. I need this functionality running in about a week, have got to find a way to make it work.

verdimrc, 2016-06-14:
+1 on the record() bug. The camera is a great product, but I've been frustrated by this bug too, and there doesn't seem to be an official channel to reach the developers?

What I notice is that record() returns false (meaning: no error occurred) even when no frame is actually written to the .SVO. Unfortunately, sleeping between consecutive record() does not completely solve the issue: on rare occasion record() can still silently do nothing.

adujardin, 2016-06-14:
Hi, 
Sorry for the inconvenience. The bug has been fixed, it will be released very soon with the next update.
@verdimrc, to reach the developers you can send an email to _support [at] stereolabs.com_, especially for issues related to the ZED SDK itself.

Nican, 2016-07-17:
When you say "fixed", do you mean alternating between grab/record modes, or being able to record and grab at the same time? 
As a developer, it is important to be able to record and use the device at the same time, for future debugging.

adujardin, 2016-07-18:
Hi,
I mean alternating (the bug described). 
We are working on being able to record and grab at the same time.

closed 2016-07-21 by adujardin