stereolabs/zed-opencv #14: ZED Depth Viewer doesn't work

dccho, 2016-08-12:
I'm running Depth Viewer in Ubuntu 14.04 desktop
When I install ZED SDK, there are two opencvs(2.4 and 3.1) in desktop
After install SDK, I remove opencv 2.4
When I run ZED Depth Viewer, there is error about missing opencv 2.4
Here is ldd of ZED Depth Viewer

ldd ZED\ Depth\ Viewer | grep opencv
	libopencv_imgproc.so.2.4 => not found
	libopencv_calib3d.so.2.4 => not found
	libopencv_highgui.so.2.4 => not found
	libopencv_core.so.2.4 => not found
	libopencv_core.so.3.1 => /usr/local/lib/libopencv_core.so.3.1 (0x00007fb2fa3bb000)
	libopencv_highgui.so.3.1 => /usr/local/lib/libopencv_highgui.so.3.1 (0x00007fb2fa1af000)
	libopencv_imgproc.so.3.1 => /usr/local/lib/libopencv_imgproc.so.3.1 (0x00007fb2f877c000)
	libopencv_calib3d.so.3.1 => /usr/local/lib/libopencv_calib3d.so.3.1 (0x00007fb2f83f1000)
	libopencv_videoio.so.3.1 => /usr/local/lib/libopencv_videoio.so.3.1 (0x00007fb2f755c000)
	libopencv_imgcodecs.so.3.1 => /usr/local/lib/libopencv_imgcodecs.so.3.1 (0x00007fb2f72c1000)
	libopencv_video.so.3.1 => /usr/local/lib/libopencv_video.so.3.1 (0x00007fb2f1d4d000)
	libopencv_flann.so.3.1 => /usr/local/lib/libopencv_flann.so.3.1 (0x00007fb2e6f4a000)

axjensen, 2016-08-12:
I experienced similar issues with the latest version listed on StereoLabs homepage, version 1.0.0c.

However, using an earlier version meanwhile might be a temporary solution, for instance version 1.0.0 found here: [https://www.stereolabs.com/download_327af3/](url)

dccho, 2016-08-15:
@axjensen Thank you~!! It works.

closed 2016-10-19 by adujardin