stereolabs/zed-opencv #16: syntax error?

hala1994, 2016-10-26:
when I run cmake ..
this error happens:
sh: 1: Syntax error: end of file unexpected (expecting ")")
then when I run make
this error happens
/home/ubuntu/zed-opencv-master/src/main.cpp: In function ‘void onMouseCallback(int32_t, int32_t, int32_t, int32_t, void*)’:
/home/ubuntu/zed-opencv-master/src/main.cpp:70:90: warning: too many arguments for format [-Wformat-extra-args]
                 printf("\n%s is too far.\n", data->name.c_str(), dist, data->unit.c_str());
                                                                                          ^
/home/ubuntu/zed-opencv-master/src/main.cpp:72:92: warning: too many arguments for format [-Wformat-extra-args]
                 printf("\n%s is too close.\n", data->name.c_str(), dist, data->unit.c_str());
                                                                                            ^
/home/ubuntu/zed-opencv-master/src/main.cpp:74:92: warning: too many arguments for format [-Wformat-extra-args]
                 printf("\n%s not avaliable\n", data->name.c_str(), dist, data->unit.c_str());
                                                                                            ^

adujardin, 2016-10-26:
Hi,
The second part about `-Wformat-extra-args` is not important, it's just warnings. The version I just committed remove this.
However the "Syntax error" is not present on a freshly cloned version of the sample in my case. Did you modify the `CMakeLists.txt` ?

If you do something like this on a terminal : 
```BASH
cd /tmp
git clone https://github.com/stereolabs/zed-opencv.git
cd zed-opencv
mkdir build; cd build; cmake ..
make
```

hala1994, 2016-10-26:
I did as you said:
cd /tmp
git clone https://github.com/stereolabs/zed-opencv.git
cd zed-opencv
mkdir build; cd build; cmake ..
make 

this is the result:

/tmp/zed-opencv$ mkdir build; cd build; cmake ..
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
sh: 1: Syntax error: end of file unexpected (expecting ")")
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "2.91.0") 
-- Found CUDA: /usr/local/cuda-6.5 (found suitable exact version "6.5") 
-- Found CUDA: /usr/local/cuda-6.5 (found version "6.5") 
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/zed-opencv/build
ubuntu@tegra-ubuntu:/tmp/zed-opencv/build$ make
Scanning dependencies of target ZED_with_OpenCV
[100%] Building CXX object CMakeFiles/ZED_with_OpenCV.dir/src/main.o
Linking CXX executable "ZED with OpenCV"
[100%] Built target ZED_with_OpenCV

hala1994, 2016-10-26:
thank you! it is running successfully

adujardin, 2016-10-26:
I don't know where the error comes from but if it's working fine, I think you can ignore it.

hala1994, 2016-10-26:
OK, Thank you !

closed 2016-10-26 by adujardin