stereolabs/zed-opencv #18: Zed opencv issues with 'init' function and 'Eigen/Core'  C++

syamprasadkr, 2016-11-16:
I am trying to apply a simple color matching algorithm to the frames coming out of Zed, in **C++**. I tried the code with image input and it worked fine. Then I changed the source file to a frame from zed using the code:
```
VideoCapture cap(0);
	if(!cap.isOpened())
	{
		cout <<" Cannot obtain video"<< endl;
		return -1;
	}
	
	Mat src; 
	cap.read(src);
	imshow("Source", src);
```
The terminal threw the following output:

```
VIDIOC_QUERYMENU: Invalid argument
VIDIOC_QUERYMENU: Invalid argument
VIDIOC_QUERYMENU: Invalid argument
```

The source frame was completely green. I tried changing `cap(0)` to `cap(-1)` and `cap (1)`. It didn't work. Then I decided to look into [this](https://www.stereolabs.com/blog/index.php/2015/06/28/zed-with-opencv/). I decided to try the [hello world ](https://www.stereolabs.com/blog/index.php/2015/07/15/hello-world/)program first. 
While running `make `, it gave me an error:

I```
n file included from /home/ubuntu/opencv_ws/zed_test/src/zed_hello.cpp:4:0:
/usr/local/zed/include/zed/Camera.hpp:73:22: fatal error: Eigen/Core: No such file or directory
 #include <Eigen/Core>
                      ^
compilation terminated.
make[2]: *** [CMakeFiles/ZED_PROJECT.dir/src/zed_hello.o] Error 1
make[1]: *** [CMakeFiles/ZED_PROJECT.dir/all] Error 2
make: *** [all] Error 2
```
Then I edited the Camera.hpp with the following lines:
```
#include <eigen3/Eigen/Core>
#include <eigen3/Eigen/Eigen>

```
Also, I changed its file permissions. Now the error says:

I```
n file included from /home/ubuntu/opencv_ws/zed_test/src/zed_hello.cpp:4:0:
/usr/local/zed/include/zed/Camera.hpp:174:21: note: sl::zed::ERRCODE sl::zed::Camera::init(sl::zed::InitParams&)
             ERRCODE init(InitParams &parameters);
                     ^
/usr/local/zed/include/zed/Camera.hpp:174:21: note:   candidate expects 1 argument, 3 provided
make[2]: *** [CMakeFiles/ZED_PROJECT.dir/src/zed_hello.o] Error 1
make[1]: *** [CMakeFiles/ZED_PROJECT.dir/all] Error 2
make: *** [all] Error 2

```
I tried changing the init statement in multiple ways: `zed->init(sl::zed::MODE::QUALITY, 0,true);`
But it didn't fix the issue. Kindly help. Thanks in advance.

sl-braun, 2016-11-16:
Hi,

Since ZED SDK v1.0, the Init() function only takes a structure as parameter : 

`zed->init(sl::zed::MODE::QUALITY, 0,true);`

becomes

```
sl::zed::InitParams param;
param.mode=QUALITY;
param.verbose=true;
param.device = 0;
zed->init(param);
```

Regarding Eigen error, it is better to call find_package(eigen3) in the cmakelist or to add /usr/include/eigen3 in the include_directories, instead of modifying the Camera.hpp.

syamprasadkr, 2016-11-16:
@sl-braun 
Thank You so much. I would like to mention the small change:
` param.mode=sl::zed::MODE::QUALITY;
`
Once again, thank you.

closed 2016-11-16 by syamprasadkr