stereolabs/zed-opencv #22: SVO File

bnascimento, 2016-11-25:
Hi guys,we are trying to load a recorded session on a SVO file on our application, but we are facing an error on camera initialization:
Relevant bit of our code:

  _sl::zed::Camera* zed;
    if ( input == "LIVE" )
      zed = new sl::zed::Camera ( mode, static_cast<float> ( fps ) );
    else
      zed = new sl::zed::Camera ( input );
    
    unsigned int firmware = zed->getZEDFirmware();
    unsigned int serial = zed->getZEDSerial();
    std::string sdkVersion = zed->getSDKVersion();
    std::cout << "Found ZED camera.\n\tserial: " << std::to_string(serial) << "\n\tfirmware: " << std::to_string(firmware) << "\n\tsdk version: " << sdkVersion << std::endl;
    
    sl::zed::InitParams params;
    params.mode = sl::zed::PERFORMANCE;
    params.verbose = true;
    
    sl::zed::ERRCODE err = zed->init ( params );
    if ( err != sl::zed::SUCCESS )
    {
        cout << "Error:" << err << endl;
        delete zed;
        return -1;
    }_ 

Where input points to the filepath of this SVO file. We get the following error message on zed->init call:

Found ZED camera.
	serial: 1825
	firmware: 700478992
	sdk version: 1.1.1
ZED SDK >> (Init) Best GPU Found : NVIDIA Tegra X1 , ID : 0
ZED SDK >> (Init) Disparity mode has been set to PERFORMANCE
ZED SDK >> (Init) Creating ZED GPU mem...
OpenCV Error: The function/feature is not implemented (Unknown/unsupported array type) in type, file /mnt/usbdev/opencv/modules/core/src/matrix.cpp, line 1879
terminate called after throwing an instance of 'cv::Exception'
  what():  /mnt/usbdev/opencv/modules/core/src/matrix.cpp:1879: error: (-213) Unknown/unsupported array type in function type

Further gdb callstack shows:

#0  0x0000007fb621a4e8 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x0000007fb621b9a0 in __GI_abort () at abort.c:89
#2  0x0000007fb63e06a8 in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/aarch64-linux-gnu/libstdc++.so.6
#3  0x0000007fb63de424 in ?? () from /usr/lib/aarch64-linux-gnu/libstdc++.so.6
#4  0x0000007fb63de470 in std::terminate() () from /usr/lib/aarch64-linux-gnu/libstdc++.so.6
#5  0x0000007fb63de6f4 in __cxa_throw () from /usr/lib/aarch64-linux-gnu/libstdc++.so.6
#6  0x0000007fb65683bc in cv::error(cv::Exception const&) () from /usr/local/lib/libopencv_core.so.3.1
#7  0x0000007fb6568574 in cv::error(int, cv::String const&, char const*, char const*, int) () from /usr/local/lib/libopencv_core.so.3.1
#8  0x0000007fb667aff0 in cv::_InputArray::type(int) const () from /usr/local/lib/libopencv_core.so.3.1
#9  0x0000007fb66d05ac in cv::Mat::copyTo(cv::_OutputArray const&) const () from /usr/local/lib/libopencv_core.so.3.1
#10 0x0000007f9e3d3930 in sl::svorw::jBQbf7SIyH::tMuEG75G5o() () from /usr/local/zed/lib/libsl_svorw.so
#11 0x0000007f9e3d1ff0 in sl::svorw::rWAL8fCthR::L4XpGGCBa(bool, unsigned char**) () from /usr/local/zed/lib/libsl_svorw.so
#12 0x0000007fb6e6dc88 in sl::zed::Camera::nextImage(bool) () from /usr/local/zed/lib/libsl_zed.so
#13 0x0000007fb6e6dda0 in sl::zed::Camera::initRectifier() () from /usr/local/zed/lib/libsl_zed.so
#14 0x0000007fb6e74a84 in sl::zed::Camera::init(sl::zed::InitParams&) () from /usr/local/zed/lib/libsl_zed.so
#15 0x0000000000422be4 in main (argc=8, argv=0x7fffffede8) at /media/ubuntu/Disk/evotegra/SignsFramework/JetsonSignRecognition_ZED/src/JetsonSignRecognition.cpp:304

Is this problem related to having OpenCV 3.1 instead of 2.4.x ?

Any hints?
Thank you for your time

sl-braun, 2016-11-25:
Hi,

On the Jetson (X1 or K1), please use the OpenCV4Tegra provided during JetPack installation (the ZED SDK is built with this version, based on OpenCV 2.4.1X and optimized by NVIDIA).

/*OB*/

bnascimento, 2016-12-06:
Ok guys, after struggling a bit to get opencv4tegra to work. I managed to open an svo file under the jetson tx1.

I am now trying to do the same on windows platform. 
Using sdk 1.1.1 right now, i managed to setup a project to compile under the delivered depedencies, everything works when we are capturing a live camera. 

However, project fails to open SVO files under windows.

Heres a snippet:

_int main( int argc, char **argv ) {

	if ( argc > 2 ) 
	{
		std::cout << "Only the path of a SVO can be passed in arg" << std::endl;
		return -1;
	}

	sl::zed::SENSING_MODE dm_type = sl::zed::FILL;
	sl::zed::Camera* zed;

	if ( argc == 1 ) // Use in Live Mode
		zed = new sl::zed::Camera( sl::zed::HD1080 );
	else // Use in SVO playback mode
		zed = new sl::zed::Camera( argv[1] );

	int width = zed->getImageSize().width;
	int height = zed->getImageSize().height;_

application crashes on zed = new sl::zed::Camera( argv[1] ); even though its a valid and existing file. I can open the video using the tools so the file itself is not corrupt.

Any hints?

sl-braun, 2016-12-06:
Hi,

Make sure you build the solution in Release mode when reading SVO files.
Since std::string does not behave the same way on Windows (in Debug and Release), the project can only be built in Release. On the upcoming 1.3 ZED SDK version, this limitation has been solved (by using const char* )

Best,
/*OB*/

bnascimento, 2016-12-07:
OK, thanks, when can we expect the 1.3 version?

sl-braun, 2016-12-07:
We will provided a release candidate end of this week or beginning of next week. You can ask the support to get access to the RC versions.
Official release will be on probably a few days (one week) after the release candidate is published.

/*OB*/