stereolabs/zed-opencv #23: ntdll.dll error using ZED sample code

eagletran, 2016-11-29:
Hi,

I am developing a system to track some colored points on a user using Zed camera. I tried the sample SVOPlayback first, but I always get an error in ntdll.dll library during runtime. My system configuration is:
Win 10 64-bit (vs 2013 redistributable installed)
Visual Studio 2013
When I open it, a pop-up window shows:
ZED SVO Playback.exe has triggered a breakpoint.

And this is the call stack:
ntdll.dll!00007ffad53073e2()	Unknown
ntdll.dll!00007ffad5307d7a()	Unknown
ntdll.dll!00007ffad52b492a()	Unknown
ntdll.dll!00007ffad52cca51()	Unknown
ntdll.dll!00007ffad524c1b8()	Unknown
msvcr120.dll!00007ffac03169d8()	Unknown
sl_zed64.dll!00007ffac67d2a24()	Unknown
>	ZED SVO Playback.exe!main(int argc, char * * argv) Line 57	C++
[External Code]

I've spent much time on this desperately. So I really really appreciate if you can help me figure out this problem.

Best regards,
Vu

sl-braun, 2016-11-29:
Hi,

Do you build in Debug or Release mode? ZED SDK is built in Release mode, therefore you should build your project in Release as well. 

Best,
/*OB*/

eagletran, 2016-11-30:
Hi,

Your suggestion is really great. it works now.

However, I come to a new problem. ZED explorer had been working perfectly, but after I've reinstalled my Window several days ago, the ZED camera didn't work correctly. I try the ZED explorer, but the frame rate is very slow and when I change the resolution to VGA (30Hz), it shows weird images as the following:
[image]

Do you have any suggestion to fix this problem?
Thank you very much.

eagletran, 2016-11-30:
Hi,

Sorry for my stupid question. I've just found the answer. It is because of my USB port, I plug the ZED into another one and it works.
any way, thank you very much for your solution of ntdll.dll error.

Best,
Vu

sl-braun, 2016-11-30:
OK. I close the issue then.

closed 2016-11-30 by sl-braun