stereolabs/zed-opencv #25: Cmake cannot link pthread properly

yahiafarghaly, 2016-12-07:
My application  require using std::thread from C++11 , when i add the library to the linker in the Cmake file by this way :

> TARGET_LINK_LIBRARIES( ${execName}
> -pthread
> 		       ${ZED_LIBRARIES} 
> 		       ${OpenCV_LIBRARIES}
> 		       ${CUDA_CUDA_LIBRARY} ${CUDA_CUDART_LIBRARY} ${CUDA_npp_LIBRARY} 
> 		       )

 the compilation is succeed but when i execute the program , i got 

> terminate called after throwing an instance of 'std::system_error'
>   what():  Enable multithreading to use std::thread: Operation not permitted
> Aborted
  
which is a famous error on the internet and i tried almost all the methods to get a rid of it but noway . 
i noticed when i did this only , it works

> TARGET_LINK_LIBRARIES( ${execName} -pthread  )
Sure i didn't link any ZED /CUDA libs in this case. 

how to solve this issue ?

sl-braun, 2016-12-08:
Hi,

You can take a look at the sample "Optimized grab" provided with the ZED SDK, that shows how to use std::thread for grabbing process. 
On the cmake file, the link is done as follow : 

```

add_definitions(-std=c++0x -O3) # -m64)
TARGET_LINK_LIBRARIES(${execName}
			pthread
                        X11
                        ${ZED_LIBRARIES}
                        ${OpenCV_LIBRARIES}
                        ${CUDA_CUDA_LIBRARY} ${CUDA_CUDART_LIBRARY} ${CUDA_npp_LIBRARY}
                    )
```

Can you also precise your OS? 

Best,
/*OB*/

axjensen, 2016-12-08:
I think it is also possible to use the following to make it compatible cross platform (only tested on Linux though):

```
cmake_minimum_required(VERSION 3.5)
project(ThreadTest)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")

find_package(ZED 1.2 REQUIRED)
find_package(Threads)

include_directories(${ZED_INCLUDE_DIRS})

link_directories(${ZED_LIBRARY_DIR})

set(SOURCE_FILES main.cpp )
add_executable(${PROJECT_NAME} ${SOURCE_FILES})

add_definitions(-std=c++0x)# -m64)

TARGET_LINK_LIBRARIES(${PROJECT_NAME}
        ${ZED_LIBRARIES}
        ${CMAKE_THREAD_LIBS_INIT}
        )

```

yahiafarghaly, 2016-12-08:
@sl-braun i did as you wrote but it didn't work , and the sample of "optimized grab" cmake file is not the same as you wrote. and when i use it for building "optimized grab" it works but when i use it for my app it doesn't.
This is my code for using thread, may something wrong ? 

```
std::vector<std::thread> threads;
  int cores = std::thread::hardware_concurrency();
  for (int i=0; i<cores; ++i) // 1 per core:
  {
    auto code = []() { 
        for(int i = 0; i< 100;i++)
        {
          std::cout << std::this_thread::get_id() <<": "<< i <<std::endl;
        }
     };
    threads.push_back( std::thread(code) );
  }
  for (std::thread& t : threads) // new range-based for:
      t.join();
```
my OS is ubuntu 14.04 on jetson TK1 , ZED SDK V1.0 
cmake version 2.8.12.2

@axjensen it doesn't work neither 

@sl-braun  , i have another question , can i install ZED SDK V1.2 without firmware update ? is running the SDKv1.2 will remove the old SDK and install the new one ?or will i need to remove old SDK manaully

axjensen, 2016-12-08:
It seems to be more a question of pthread usage than zed or OpenCV, so from the following example (working on a similar platform to yours) is without any reference to ZED or OpenCV:

CMakeLists.txt:

```
cmake_minimum_required(VERSION 3.6)
project(pthread_test)

set(CMAKE_CXX_STANDARD 11)
set(SOURCE_FILES main.cpp)
set(THREADS_PREFER_PTHREAD_FLAG ON)
find_package(Threads REQUIRED)

add_executable(pthread_test ${SOURCE_FILES})
target_link_libraries(${PROJECT_NAME} Threads::Threads)
```

main.cpp:

```
#include <iostream>
#include <thread>
#include <vector>

int main() {
    std::vector<std::thread> threads;
    int cores = std::thread::hardware_concurrency();
    for (int i = 0; i < cores; ++i) // 1 per core:
    {
        auto code = []() {
            for (int i = 0; i < 100; i++) {
                std::cout << std::this_thread::get_id() << ": " << i << std::endl;
            }
        };
        threads.push_back(std::thread(code));
    }
    for (std::thread &t : threads) // new range-based for:
        t.join();

    return 0;
}
```

closed 2022-04-27 by github-actions[bot]