stereolabs/zed-opencv #26: enableRecording throws access violation error

ShacharWeis, 2017-01-20:
Hey all,

I can't get recording to work. I follow the sample, create a camera, init it, and when I call enableRecording it always throws an Access Violation error.

Any ideas?

sl-braun, 2017-01-21:
Hi,

Can you indicate the ZED SDK version you are using?

Best,
OB.

ShacharWeis, 2017-01-21:
I am using 1.2.0

I have found that it only crashes when compiled with "Debug". With "Release" it works.

ShacharWeis, 2017-01-21:
Everything about record and playback is broken in DEBUG. Calling m_camera->getSVONumberOfFrames() returns the correct value in RELEASE, but in DEBUG it returns a large negative value.

sl-braun, 2017-01-23:
You need to use Release mode in Windows when using SVO files in the 1.2.
This limitation has been solved in upcoming release.

OB

ShacharWeis, 2017-01-23:
Thanks.
Do you know when the upcoming release is?

I am writing some complex algorithms, and I need to test them on known and unchanging input (hence, a recording) and I really need the DEBUG mode.

jgirado, 2017-04-18:
Hi,

I have the exact same problem, enableRecording() crashing in Debug mode (VS 2015). I'm using SDK 2.0.0 downloaded a week ago). I can't debug my code because because of this. Any plans to fix it?