stereolabs/zed-opencv #27: SDK 1.2 support on TX1 (jetpack 2.3.1)

NotInControl, 2017-02-11:
Trying to build opencv example on Jetson TK1 running jetpack 2.3.1, simply pulling the github repo (which works on my Tegra TK1, running Jetpack 2.3.1 fails on TX1 with missing opencv lib). This is a vanilla machine, just flashed with jetpack, no other modifications. 

ubuntu@tegra-ubuntu:~$ git clone https://github.com/stereolabs/zed-opencv.git
Cloning into 'zed-opencv'...
remote: Counting objects: 47, done.
remote: Total 47 (delta 0), reused 0 (delta 0), pack-reused 47
Unpacking objects: 100% (47/47), done.
Checking connectivity... done.
ubuntu@tegra-ubuntu:~$ cd zed-opencv/
ubuntu@tegra-ubuntu:~/zed-opencv$ ls
CMakeLists.txt  LICENSE  README.md  src
ubuntu@tegra-ubuntu:~/zed-opencv$ mkdir build
ubuntu@tegra-ubuntu:~/zed-opencv$ cd build/
ubuntu@tegra-ubuntu:~/zed-opencv/build$ cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda-8.0 (found suitable exact version "8.0") 
-- Found CUDA: /usr/local/cuda-8.0 (found version "8.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/zed-opencv/build
ubuntu@tegra-ubuntu:~/zed-opencv/build$ make
Scanning dependencies of target ZED_with_OpenCV
[ 50%] Building CXX object CMakeFiles/ZED_with_OpenCV.dir/src/main.o
[100%] Linking CXX executable "ZED with OpenCV"
/usr/bin/ld: cannot find -lopencv_dep_cudart
collect2: error: ld returned 1 exit status
CMakeFiles/ZED_with_OpenCV.dir/build.make:134: recipe for target 'ZED with OpenCV' failed
make[2]: *** [ZED with OpenCV] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ZED_with_OpenCV.dir/all' failed
make[1]: *** [CMakeFiles/ZED_with_OpenCV.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

sl-braun, 2017-02-11:
Hi,

you need to call cmake with -DCUDA_USE_STATIC_CUDA_RUNTIME=false to avoid opencv linking error.

Best,
OB.

MichaelX99, 2017-04-27:
Is there a way to add -DCUDA_USE_STATIC_CUDA_RUNTIME=false into the cmake file so we can only call cmake from the terminal instead of always passing it this argument?

closed 2022-04-27 by github-actions[bot]