stereolabs/zed-opencv #28: Multiple main files after cmake operation

Patelvishal95, 2017-02-13:
Hey i ran cmake on your cloned github project. When i build the project after changing the build mode to release. i get the following error code.
Error	2	error LNK1169: one or more multiply defined symbols found	D:\zed-opencv-master\src\Release\ZED with OpenCV.exe	ZED_with_OpenCV
Error	1	error LNK2005: main already defined in CMakeCXXCompilerId.obj	D:\zed-opencv-master\src\main.obj	ZED_with_OpenCV
there seems to be some error in cmake that i am not able to locate.

madajimmy, 2018-11-05:
You have two main functions in your project.
Delete or comment main function in CMakeCXXCompilerId.obj file and it should be fine.
It locates at the bottom of file.

windwolf, 2020-09-02:
I got the same problem...

crysty936, 2022-01-14:
If anyone ever stumbles upon this in the future, it's because most probably you have globbed everything under the CMAKE_CURRENT_LIST_DIR as source files, that includes everything in the Build Folder, which contains source files that should definitely not be in your project. Consider filtering those files from your project.

closed 2022-04-27 by github-actions[bot]