stereolabs/zed-opencv #31: Example of recording rectified stereo footage?

mtee, 2017-03-06:
I have seen that the function grab() returns rectified images from the ZED camera. Is there any example of saving these images into a video file? The ZED Explorer is saving a lot of auxiliary data in the SVO file, which I don't need in this particular application, so I am looking for a way to get a lighter and more stable recording (720p at 60fps), than with the ZED Explorer. Is the opencv-wrapper the place to start? Will it have less frame drops than the ZED Explorer?

sl-braun, 2017-03-06:
If you are using the ZED SDK 1.2, the best workflow is the following : 

- record a SVO file with ZED Explorer, using a lossy compression (fatest, less frame drop)
- Use SVO Converter sample to convert svo to avi file. It will give you recitifed images.

Best,
OB

mtee, 2017-03-20:
Thank you for your reply @sl-braun, one more question regarding to the recording: 
which calibration parameters should I use with the rectified footage I acquire the way you proposed? Should I use the cx, cy, fx, fy of the left camera or right?

sl-braun, 2017-03-20:
Rectified images means that they are processed (digitally) as if they were coming from a absolutely perfect stereo camera : it means identical cameras and perfectly aligned.
Therefore, since fx,fy,cx,cy defines the camera matrix once rectified, left camera == right camera (for those values). Doesn't matter if you take the left or the right.

mtee, 2017-03-20:
Well yes, both left and right images are undistorted and aligned, but is the right image aligned to the left one (that is, the cx and cy of the left one are preserved), or the other around? In my understand the undistortion does not affect the cx, cy, fx, fy parameters. However the alignment of images to each other does.