stereolabs/zed-opencv #32: The delay problem of the opencv sample

liyongle1984, 2017-04-19:
Hi, I use the opencv sample to get the captured image and depth image. But I found the image stream has a little delay about 100ms while using the sample. If I use the zed tool "ZED Depth Viewer" to show the captured image and depth image, there is almost no delay.  Can anybody tell me the reason and how to fix it? Thanks.

sl-braun, 2017-04-19:
Hi,

The difference between OpenCV sample and ZED Depth Viewer results in the way images are displayed : 

- With OpenCV, images are taken back from the GPU and then displayed with opencv imshow()
- ZED Depth Viewer uses directly GPU images and opengl-cuda interoperability (no GPU->CPU)

the delay you have might come from that, especially if you are on a jetson board (where GPU->CPU are not recommended). To have an code similar to Depth Viewer, you can check examples/other/opengl gpu interop sample.

Best,
OB

liyongle1984, 2017-04-20:
Thanks very much.

Ted-EE, 2020-07-06:
https://github.com/stereolabs/zed-opencv/issues/32#issuecomment-295139055
@obraun-sl Hi, I met similar problem when doing some work on real sense D415, but the link you provided cannot be viewed now, I appreciate it if you give me some suggestions on the problem or provide a new link 

Thank!

obraun-sl, 2020-07-06:
here is the link : https://github.com/stereolabs/zed-opengl
Or with Qt as opengl support (QGLWidget) : https://github.com/obraun-sl/qtZEDViewer

Ted-EE, 2020-07-06:
Thanks for your replying! But my code is in Python, and my depth camera is Realsense D415, could my problem be solved using this link?

obraun-sl, 2020-07-06:
The links are for the ZED Camera. You might find some ideas for the realsense (best is to open an issue on their repo...)
for python and opengl, you can use PyOpenGL

Ted-EE, 2020-07-06:
I will look it and may find some ideas, thanks for your help!

closed 2022-04-27 by github-actions[bot]