stereolabs/zed-opencv #34: Rectification of Images captured using ZED Explorer

shanuv, 2017-05-25:
My laptop doesn't have Cuda installed. Hence, I captured some images using the ZED explorer in Windows and saved the png files. When I run the depth estimation code on the images, I get very poor results and I think it is because the images are not rectified whereas the depth code runs on rectified images only. Should I use opencv now to rectify the images or is there a better alternative available using the ZED SDK? Thanks in advance for the help!

WASCHMASCHINE, 2017-05-29:
Why can't you install CUDA?
If I remember correctly, the images are rectified in the ZED explorer preview. If you only use the ZED as a fixed stereo camera rig, you will lose most of the performance it offers. If you still want to use the ZED camera, you can try using OpenCV to get images from the cameras and then A) calibrate them yourself with a checkerboard or B) use the provided calibration. Then you can compute the rectified images and run your own stereo matching (e.g. LIBELAS, or SGM GPU -- search github).

Here are the docs:
http://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html

LUCASLLA, 2018-04-03:
Hi @shanuv , have you discovered if the images you took from the ZED Explorer are rectified or you need to rectify them?

closed 2022-04-27 by github-actions[bot]