stereolabs/zed-opencv #39: ZED / focal length ; Size of sensors  ; Size of a Photosite

gauthier-duponchel, 2017-06-28:
In order to calibrate each camera of the ZED using MicMAc, I need those informations : 
- focal length (mm) 
- sensor size ( px . px )
- photosite size (m) 

Gauthier

WASCHMASCHINE, 2017-06-28:
Open the ZED Explorer -> Settings -> Calibration, or just look at the calibration file of your camera. I think you can derive your values from there for each setting.

gauthier-duponchel, 2017-06-28:
In the SNxxxx.conf file [calibration file], I can find fx and fy which are [I think] the projection of the optical center on the sensor but I do not see the focal length in meters which should be a kind of fz.

Concerning the sensor size, there is unfortunately no information concerning the size in m.

Gauthier

adujardin, 2017-06-28:
Hi,
Here is the pixel size by resolution:
HD2K (2208x1242) : 0.002mm
HD1080 (1920x1080) : 0.002mm
HD720 (1280x720) : 0.004mm
WVGA (672x376) : 0.008mm

gauthier-duponchel, 2017-06-28:
Thank you!!

What about the focal length ?
What are fx & fy  in the SNxxxx.conf file / focal length ?

Gauthier

gauthier-duponchel, 2017-06-28:
Thank you!

What about the focal length ??

In the SNxxxx.conf file I can find fx and fy but I think they are the
coordinates of the optical center projection on the sensor.
What I search is the focal length (fz if you see what I neam).

Gauthier

adujardin, 2017-06-28:
The focal length is 2.8mm.

You can also compute it from the focal in pixels in the SNxxxx.conf :
- `f_mm = (fx+fy) / 2 * pixel_size_mm`

gauthier-duponchel, 2017-06-28:
OK!
Thank you again, Gauthier.

closed 2017-06-28 by adujardin