stereolabs/zed-opencv #40: ZED camera has access violation error

nyanmn, 2017-07-06:
I used ZED camera to capture 3-D images.
At the api grab(), I have access violation error.
I can run the sample exe file for depth viewer, just that interfacing from my code has issue. That happened at my industrial PC.
The same code I used at my development PC and no issue.
What could be wrong?
Thanks

WASCHMASCHINE, 2017-07-06:
Post more details about your setup and share some code!

nyanmn, 2017-07-06:
I have 
```
void BaggageSoln::startZED()
{
	debugtext << "inside startZED" << endl;
	// Start the thread for grabbing ZED data
	has_data = false;
	debugtext << "inside startZED, before thread" << endl;
	zed_callback = std::thread(&BaggageSoln::run, this);
	debugtext << "inside startZED, thread is called" << endl;
	//Wait for data to be grabbed
	while (!has_data)
		sl::sleep_ms(1);
}
Then inside run(); I have
void BaggageSoln::run() {
	debugtext << "inside run" << endl;
	while (!stop_signal) {
		debugtext << "inside run, before grab" << endl;
		ERROR_CODE e = zed.grab(sl::SENSING_MODE_STANDARD);
		debugtext << "inside run, before grab" << endl;
		if (e == SUCCESS) {
			mutex_input.lock(); // To prevent from data corruption
			zed.retrieveMeasure(data_cloud, MEASURE_XYZRGBA);//MEASURE_XYZ //MEASURE_XYZRGBA
			zed.retrieveImage(leftImg_sl, sl::VIEW_LEFT);
			cv::Mat crop;
			cv::Mat temp = cv::Mat(leftImg_sl.getHeight(), leftImg_sl.getWidth(), CV_8UC4, leftImg_sl.getPtr<sl::uchar1>(sl::MEM_CPU));
			cv::cvtColor(temp, crop, CV_RGBA2GRAY);
			crop(cv::Rect(xpt, ypt, rectl, rectw)).copyTo(leftImg);	
			for (int i = 3; i < 7; i = i + 2)
			{
				GaussianBlur(leftImg, leftImg, cv::Size(i, i), 0, 0);
			}				
			crop.release();
			temp.release();
			mutex_input.unlock();
			has_data = true;			
		}
}
The error always happen at
`ERROR_CODE e = zed.grab(sl::SENSING_MODE_STANDARD);`

Since I make the C++ dll and run from C# exe program, the error at C# output is 
exe has existed with code (0XC0000005) ACCESS VIOLATION

What could be the issue?
I can run sample programs from ZED SDK/tools/ and ZED SDK/app.

But can't run from ZED SDK/samples/bin, those exe program gives me error as error:usb low bandwidth.

nyanmn, 2017-07-06:
I check error number and it is 6, that means ERROR_CODE_LOW_USB_BANDWIDTH. But why I can run those exe files from ZED SDK/tools/ and ZED SDK/app.

nyanmn, 2017-07-06:
access violation error is because of camera is not opened with ERROR_CODE_LOW_USB_BANDWIDTH error.

sl-braun, 2017-07-06:
Hi,

This error happens when a lot of corrupted frames are detected.
Either try to reduce the FPS in InitParameters to reduce the data size, or try to use a powered USB hub to give more power to the camera.

PS : If possible, try to give some information about your PC (Win7,8,10, Linux, Jetson....)

/*OB*/

nyanmn, 2017-07-06:
Yes now I changed to 60 fps and it works. Thank you.

closed 2017-07-11 by nyanmn