stereolabs/zed-opencv #41: opencv 3.1

lioyijie, 2017-07-07:
I want to install opencv 3.1 as the zed sdk camera seems to only support opencv 3.1.
However, many links that I search for seems to take the sources from https://github.com/opencv/opencv 
and even though the website which is this "http://docs.opencv.org/3.1.0/d7/d9f/tutorial_linux_install.html" says that it is installing opencv 3.1, but when I finish the steps in the website, it appears to be opencv 3.3.
Does anyone know how to install opencv 3.1 instead of opencv 3.3? Thanks

GPrathap, 2017-07-08:
1.  `git clone https://github.com/opencv/opencv.git`
2.  `cd opencv`
3.  `git checkout 3.1.0`
4.  `mkdir build `
5.  `cd build`
6.  `cmake .. //configure which you want to install`
7.  `make -j4`
8.  `sudo make install`

lioyijie, 2017-07-11:
@GPrathap 
Thanks for the help, but when I followed the steps, there's all this error at step number 7

 /home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp:120:54: error: ‘NppiGraphcutState’ has not been declared
     typedef NppStatus (*init_func_t)(NppiSize oSize, NppiGraphcutState** ppState, Npp8u* pDeviceMem);
                                                      ^
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp:135:18: error: ‘NppiGraphcutState’ does not name a type
         operator NppiGraphcutState*()
                  ^
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp:141:9: error: ‘NppiGraphcutState’ does not name a type
         NppiGraphcutState* pState;
         ^
In file included from /home/gn5/opencv/build/modules/cudalegacy/precomp.hpp:75:0:
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp: In constructor ‘{anonymous}::NppiGraphcutStateHandler::NppiGraphcutStateHandler(NppiSize, Npp8u*, {anonymous}::init_func_t)’:
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp:127:39: error: ‘pState’ was not declared in this scope
             nppSafeCall( func(sznpp, &pState, pDeviceMem) );
                                       ^
/home/gn5/opencv/modules/core/include/opencv2/core/private.cuda.hpp:165:52: note: in definition of macro ‘nppSafeCall’
 #define nppSafeCall(expr)  cv::cuda::checkNppError(expr, __FILE__, __LINE__, CV_Func)
                                                    ^
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp: In destructor ‘{anonymous}::NppiGraphcutStateHandler::~NppiGraphcutStateHandler()’:
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp:132:43: error: ‘pState’ was not declared in this scope
             nppSafeCall( nppiGraphcutFree(pState) );
                                           ^
/home/gn5/opencv/modules/core/include/opencv2/core/private.cuda.hpp:165:52: note: in definition of macro ‘nppSafeCall’
 #define nppSafeCall(expr)  cv::cuda::checkNppError(expr, __FILE__, __LINE__, CV_Func)
                                                    ^
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp:132:49: error: ‘nppiGraphcutFree’ was not declared in this scope
             nppSafeCall( nppiGraphcutFree(pState) );
                                                 ^
/home/gn5/opencv/modules/core/include/opencv2/core/private.cuda.hpp:165:52: note: in definition of macro ‘nppSafeCall’
 #define nppSafeCall(expr)  cv::cuda::checkNppError(expr, __FILE__, __LINE__, CV_Func)
                                                    ^
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp: In function ‘void cv::cuda::graphcut(cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::Stream&)’:
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp:174:51: error: ‘nppiGraphcutGetSize’ was not declared in this scope
     nppSafeCall( nppiGraphcutGetSize(sznpp, &bufsz) );
                                                   ^
/home/gn5/opencv/modules/core/include/opencv2/core/private.cuda.hpp:165:52: note: in definition of macro ‘nppSafeCall’
 #define nppSafeCall(expr)  cv::cuda::checkNppError(expr, __FILE__, __LINE__, CV_Func)
                                                    ^
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp:182:61: error: ‘nppiGraphcutInitAlloc’ was not declared in this scope
     NppiGraphcutStateHandler state(sznpp, buf.ptr<Npp8u>(), nppiGraphcutInitAlloc);
                                                             ^
In file included from /home/gn5/opencv/build/modules/cudalegacy/precomp.hpp:75:0:
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp:191:146: error: ‘nppiGraphcut_32s8u’ was not declared in this scope
             static_cast<int>(terminals.step), static_cast<int>(leftTransp.step), sznpp, labels.ptr<Npp8u>(), static_cast<int>(labels.step), state) );
                                                                                                                                                  ^
/home/gn5/opencv/modules/core/include/opencv2/core/private.cuda.hpp:165:52: note: in definition of macro ‘nppSafeCall’
 #define nppSafeCall(expr)  cv::cuda::checkNppError(expr, __FILE__, __LINE__, CV_Func)
                                                    ^
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp:196:146: error: ‘nppiGraphcut_32f8u’ was not declared in this scope
             static_cast<int>(terminals.step), static_cast<int>(leftTransp.step), sznpp, labels.ptr<Npp8u>(), static_cast<int>(labels.step), state) );
                                                                                                                                                  ^
/home/gn5/opencv/modules/core/include/opencv2/core/private.cuda.hpp:165:52: note: in definition of macro ‘nppSafeCall’
 #define nppSafeCall(expr)  cv::cuda::checkNppError(expr, __FILE__, __LINE__, CV_Func)
                                                    ^
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp: In function ‘void cv::cuda::graphcut(cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::GpuMat&, cv::cuda::Stream&)’:
/home/gn5/opencv/modules/cudalegacy/src/graphcuts.cpp:246:52: error: ‘nppiGraphcut8GetSize’ was not declared in this scope
     nppSafeCall( nppiGraphcut8GetSize(sznpp, &bufsz) );
[... cut ...]
modules/cudalegacy/CMakeFiles/opencv_cudalegacy.dir/build.make:242: recipe for target 'modules/cudalegacy/CMakeFiles/opencv_cudalegacy.dir/src/graphcuts.cpp.o' failed
make[2]: *** [modules/cudalegacy/CMakeFiles/opencv_cudalegacy.dir/src/graphcuts.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:9173: recipe for target 'modules/cudalegacy/CMakeFiles/opencv_cudalegacy.dir/all' failed
make[1]: *** [modules/cudalegacy/CMakeFiles/opencv_cudalegacy.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

GPrathap, 2017-07-11:
Yeap, There is an issue in OpenCV 3.1 which related to Cuda compilation. This is the fix for that. https://github.com/opencv/opencv/pull/6510

Edit the OpenCV source code according to this  https://github.com/opencv/opencv/commit/af64ecdf25e450785c9abf29cfd2085c01d027fb and do the build. I also had same issue while building Opencv 3.1. This fix works for me.

guruvishnuvardan, 2018-02-10:
Hi Prathap, Hi Folks,
I tried to change the code of graphcuts.cpp to the following, buts readonly

+#if !defined (HAVE_CUDA) || defined (CUDA_DISABLER) || (CUDART_VERSION >= 8000)

Can you please let me know what's the issue here is

closed 2022-04-27 by github-actions[bot]