stereolabs/zed-opencv #43: Using ZED camera for object with glossy surface at outdoor

nyanmn, 2017-07-11:
I use ZED camera to measure object dimension for length x width x height.
The camera is on TOP at about 1.6m height and the object is below.
The issue is at outdoor, bright light like sunlight makes problem for objects with glossy surface.
Can't reconstruct complete 3D shape for the whole object. Sometimes at the middle has black hole and the whole object is not reconstructed in 3D shape.
Like that environment, are there any parameters to fine tune? Say I can detect that the whole 3D shape is not reconstructed then tune that parameters and make better reconstructions.
What are the parameters need to look at for (1) 3D reconstruction is not complete for the whole object (2) what parameters to be tuned so that I have better quality of reconstruction?
Thanks

WASCHMASCHINE, 2017-07-11:
This is an inherent stereo problem. What 3D shapes are you talking about? And are you using ZEDfu?

You could process the data regularize the reconstruction with some knowledge of the prior shape, e.g. for houses with windows you could enforce some planarity constraint to fill holes at window surfaces.

nyanmn, 2017-07-11:
Baggages at airport. Airplane landed, sometimes the place is quite bright with no shade. Some baggages are glossy type.

mindmad, 2019-04-29:
Hello, @nyanmn  I am to trying to use zed camera to measure object dimension,
If you don't mind can you show me how or at least any kind of help
I AM STILL A BEGINNER so anything is helpful.
I will be waiting to hear from you

nyanmn, 2019-04-29:
What is ur difficulty?
Just install library and can start use

mindmad, 2019-04-30:
> What is ur difficulty? Just install library and can start use
> […](#)

well, I am a beginner. so any guidelines of any kind will help

mindmad, 2019-05-04:
@nyanmn  please any help, I have installed the zed library and use them but  I don't have a clear idea how to measure object size with them.

closed 2022-04-27 by github-actions[bot]