stereolabs/zed-opencv #47: How to get the GpuMat of depth image?

DreamsZM, 2017-08-28:
I wish to get the GpuMat of depth image,what should I do?

WASCHMASCHINE, 2017-08-28:
Look at the sample code: https://github.com/stereolabs/zed-examples

DreamsZM, 2017-08-29:
@WASCHMASCHINE after get zed.retrieveMeasure(gpu_depth, sl::MEASURE_DEPTH, sl::MEM_GPU),how to transform the cv::gpu::GpuMat?

sl-braun, 2017-08-29:
Use cudaMemcpy to copy the GPU data of sl::Mat (using getPtr<sl::float1>(MEM_GPU)) to the gpu data of the gpuMat (.data)

DreamsZM, 2017-08-29:
`sl::Mat inputDepthMat;
zed.retrieveImage(inputDepthMat, sl::VIEW_DEPTH, sl::MEM_GPU);
......
cv::gpu::GpuMat inputDepthMat_GPU;
cudaMemcpy2D(inputDepthMat_GPU.data, sizeof(float), inputDepthMat.getPtr<sl::uchar1>(sl::MEM_GPU), sizeof(float), 
		         inputDepthMat.getWidth(), inputDepthMat.getHeight(), cudaMemcpyDeviceToDevice);`

@sl-braun @WASCHMASCHINE

liprin1129, 2018-07-05:
I did like this below, but not working...
`		cudaMemcpy2D(
				this->_gpuMat.data,
				this->_gpuMat.step * sizeof(float3),
				this->_zedMat.getPtr<sl::uchar1>(sl::MEM_GPU),
				this->_zedMat.getStepBytes(sl::MEM_GPU),
				this->_zedMat.getWidth() * sizeof(sl::uchar4),
				this->_zedMat.getHeight(),
				cudaMemcpyDeviceToDevice
				);`

closed 2022-04-27 by github-actions[bot]