stereolabs/zed-opencv #48: Function symbol conflict in libsl_core.so and libopencv_core.so

sulpha, 2017-08-29:
I'm use both ZED SDK 2.1.2 and OpenCV 3.2.

My code is as follow :
```
void saveMat(const string& filePath, const Mat& mat) {
  cv::FileStorage fs(filePath, cv::FileStorage::WRITE);
  bool isOk = myfs.isOpened();
  std::cout << "isOpened: " << isOk << std::endl;
  ...
}
```
I've found some strange behavior:
The bool variable isOk's value is not 0 or 1, it's a random number less than 255.

Debug use gdb, I found:
cv::FileStorage use cvOpenFileStorage() function internal, but this function both exists in libsl_core.so and libopencv_core.so, and last call libsl_core.so's cvOpenFileStorage() function, I don't know what's happend in libsl_core.so's cvOpenFileStorage() that the return bool's value is not 1 or 0. 

The ld linker searches and processes libraries and object files in the order they are specified. 

Use nm command :
nm -D --defined-only /usr/local/zed/lib/libsl_core.so
I've find many OpenCV's function, why?
```
000000000034f140 T cvMulTransposed
00000000001eaae0 T cvNextGraphItem
0000000000224660 T cvNextNArraySlice
00000000001eb770 T cvNextTreeNode
000000000031eb10 T cvNorm
00000000001c5890 T cvNormalize
00000000005da940 T cvNormalizeHist
0000000000210b70 T cvNot
00000000002a3620 T cvNulDevReport
00000000002bd2b0 T cvOpenFileStorage
0000000000211690 T cvOr
0000000000212ce0 T cvOrS
000000000034e580 T cvPerspectiveTransform
000000000047e580 T cvPointPolygonTest
0000000000572b80 T cvPointSeqFromMat
00000000002cc790 T cvPolarToCart
00000000006556f0 T cvPolyLine
00000000000b5dd0 T cvPOSIT
00000000002cdbc0 T cvPow
```

[image]

[image]

sl-braun, 2017-08-29:
Hi,

New version of  ZED SDK uses some OpenCV functions internally but they are protected so it should not interfere with OpenCV standard library.
Anyway you may try to put the {OPENCV_LIBRARIES} before any other libs in the cmakelist file to check if it works better.

sulpha, 2017-08-29:
Put the {OPENCV_LIBRARIES} before any other libs  not works for me. The generated CMakeFiles/{PROJECT}.dir/link.txt file always put libsl_core.so before libopencv_core.so.

sulpha, 2017-08-29:
Is it better put all OpenCV about functions in a seperated namespace.

closed 2022-04-27 by github-actions[bot]