stereolabs/zed-opencv #52: Which version of openCV to use with the latest zed sdk 2.3.2

vijay609, 2018-02-28:
Trying to use the zed Camera on TX2 and on a standard linux box. Do you have steps documenting how to install (or build from source) openCV and all the dependencies? Also what version of openCV is recommended. This page doesn't talk about openCV installation steps
https://docs.stereolabs.com/integrations/opencv/getting-started/

adujardin, 2018-02-28:
Hi,
Since the ZED SDK doesn't depend on OpenCV anymore, you're free to use any recent version. 

Either install it from the package manager or build from source, something like OpenCV > 3.2 to get the best performance possible.
You can follow [this tutorial](https://docs.opencv.org/trunk/d7/d9f/tutorial_linux_install.html) to build OpenCV

vijay609, 2018-02-28:
Ok. Thanks for the quick reply.

closed 2018-02-28 by vijay609

pavelpetrovic, 2020-04-24:
It would be nice to have a note about this in the ZED documentation, I can't find a word about it. Now there is ZED SDK 3, and I do not know how to guess which version of OpenCV I need in order to build the opencv samples. As building from sources takes some time, trial and error process is not something extremely enjoyable.

obraun-sl, 2020-04-27:
Hi,
ZED SDK 3.0 does not use OpenCV as a dependency. Therefore you can use any OpenCV 3 or 4 version in the samples. Even OpenCV2 might work with small API changes.

pavelpetrovic, 2020-04-27:
Thank you. I tried the most recent one, and it worked smooth. Given the fact that the api change a bit from time to time, I thought the information you provided here now would have made me even more happy if I would have found it in the samples' README file at the time I needed it. Have a nice day.