stereolabs/zed-opencv #57: CMAKE does not work

wangandxiao, 2018-05-16:
I tried to use cmake to build the project. However, the following error messages shows. 
CMake Warning at C:/Program Files (x86)/ZED SDK/dependencies/opencv_3.1.0/OpenCVConfig.cmake:166 (message):
  Found OpenCV Windows Pack but it has no binaries compatible with your
  configuration.

  You should manually point CMake variable OpenCV_DIR to your build of OpenCV
  library.
Call Stack (most recent call first):
  CMakeLists.txt:13 (find_package)

CMake Error at CMakeLists.txt:13 (find_package):
  Found package configuration file:

    C:/Program Files (x86)/ZED SDK/dependencies/opencv_3.1.0/OpenCVConfig.cmake

  but it set OpenCV_FOUND to FALSE so package "OpenCV" is considered to be
  NOT FOUND.
I tried other ZED project without using opencv and they all work. Anyone knows how to fix this? Thanks

obraun-sl, 2018-05-16:
Hi,

The OpenCV version provided by the SDK only contains binaries compatible with VS2015 x64 (see the vc14 folder).
Either use VS2015 x64 when you choose the compiler with cmake, or install your own opencv version ( with VS2017 binaries for example)

SumairaManzoor, 2018-09-06:
Hi, 
I am newbie.
I am also facing the same issue, after installing OpenCV 3.4.3 win pack.

---- Windows 10
----- Visual Studio 15 2017 Win64
----- CUDA  9.2
---- OpenCV 3.4.3 win pack

@obraun-sl  
To solve this issue:  How can I make VS2017 binary compatible according to your suggestion?
Thank you

=============
ERROR
=============
Found CUDA: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v9.2 (found suitable exact version "9.2") 
OpenCV ARCH: x64
OpenCV RUNTIME: 
OpenCV STATIC: ON
CMake Warning at C:/Program Files (x86)/ZED SDK/dependencies/opencv_3.1.0/OpenCVConfig.cmake:166 (message):
  Found OpenCV Windows Pack but it has no binaries compatible with your
  configuration.

  You should manually point CMake variable OpenCV_DIR to your build of OpenCV
  library.
Call Stack (most recent call first):
  CMakeLists.txt:13 (find_package)

CMake Error at CMakeLists.txt:13 (find_package):
  Found package configuration file:

    C:/Program Files (x86)/ZED SDK/dependencies/opencv_3.1.0/OpenCVConfig.cmake

  but it set OpenCV_FOUND to FALSE so package "OpenCV" is considered to be
  NOT FOUND.

obraun-sl, 2018-09-06:
Cmake is still looking at OpenCV in the ZED SDK dependencies folder. That's why it still does not find opencv proper binaries.

* Make sure first you delete the cache for cmake. (File >> Delete cache)
* Click on configure, select VS2017.
* Look at the OPENCV_DIR variable in the cmake entry. It must be your opencv 3.4 installation. If it's not the case , then change it to the correct one and click on configure again.

SumairaManzoor, 2018-09-06:
Thank  You
Would you please guide me 
Do I have to edit the OpenCVConfig.cmake file in OpenCV_3.1.0 folder for chaning OPENCV_DIR variable?

As this file can not be edited directly on windows 10. So I would have to install any cmake editor to change the variable value in this file.

SumairaManzoor, 2018-09-07:
Thank You
Yes, I am using windows 10 and I have installed openCV in C:/Program Files(x86)/ZED SDK/dependencies/opencv_3.4.3 now.
When I select OpenCV_CONFIG_PATH through cmake GUI. It asks for File Name. I don't know which file should I specify here. So I gave every file in bin (opencv_3.4.3/x64/vc15/bin/) one by one. 
But it shows the same error and also resets the path of OpenCV_CONFIG_PATH as C:/Program Files(x86)/ZED SDK/dependencies/opencv_3.1.0 when I click "Configure" button.

Although, I have set the environmentanal variables as well.

Still facing the same problem.. So need your help.
Thanks

Error screenshot 
----------------------------
It asks for file name
[image]

--------------------------------

Gives the same error everytime
[image]

-----------------
and resets the path to opencv_3.1.0  again
[image]

obraun-sl, 2018-09-07:
You need to put OpenCVConfig.cmake for OpenCV_CONFIG_PATH. 
If OpenCV is properly installed, it should be located in C:/Program Files(x86)/ZED SDK/dependencies/opencv_3.4.3/ on your PC.

SumairaManzoor, 2018-09-07:
Thank You.
I did it but still the problem is there. 
Although, OpenCV is properly installed because OpenCVConfig.cmake is located in C:/Program Files(x86)/ZED SDK/dependencies/opencv_3.4.3/. 

[image]

obraun-sl, 2018-09-07:
* Remove opencv3.1.0 folder from dependencies to avoid conflicts (or zip it and remove folder if you want to keep a trace)
* Select opencv3.4/OpenCVConfig.cmake in OpenCV_CONFIG_PATH and press configure.
* You may still have an error but it should create a field OpenCV_DIR. Set the value as your opencv3.4 path. Press configure again. If you have "configure done" at the end, you can generate the solution.

SumairaManzoor, 2018-09-07:
In this case, the problem is if I remove opencv3.1.0 folder from dependencies  then it does not show OPENCV in cmake GUI to reset OpenCV_CONFIG_PATH and OpenCV_DIR. Also gives error

[image]

If I install Visual Studio 2015 and use OpenCV_3.1.0, also I download SDK again after installing VS 2015 then what do you think, this openCV_3.1.0 problem would be solved?  

May be this issue is due to VS 2017 therefore cmake also gives this error

[image]

As you can see these 2 errors 
[image]

So, what would you suggest? should I install VS 2015 and reinstall ZED SDK?

obraun-sl, 2018-09-07:
On your first picture, the OpenCV_Dir appears but in ungroupped entries. Try to set it to see what it gives.

On my side, I had no problem using VS2017 with ZED SDK 2.5 and opencv 3.4 (your configuration).

SumairaManzoor, 2018-09-08:
Thank You 
This error is fixed now by setting OpenCV_Dir inCMakeLists.txt as: SET(OpenCV_DIR "C:/Program Files (x86)/ZED SDK/dependencies/opencv_3.4.3") 

It shows 
Configurating Done
Generating Done

**But it also gives the message  "You might need to add  C:\Program Files (x86)\ZED SDK\dependencies\opencv_3.4.3\x64\vc15\bin" to your path to be able to run your application.**

**Therefore, I am facing another issue now. When I run Zed Camera Control Project in Visual Studio 2017. It gives error "Camera Failed to Setup". In Debug, it also shows "The program '[1096] ZED_Camera_Control.exe' has exited with code 1 (0x1)."**

I tried to solve this  issue by adding SET(OpenCV_CONFIG_PATH "C:/Program Files (x86)/ZED SDK/dependencies/opencv_3.4.3/x64/vc15/bin") in CMakeLists.txt . But still it displays the same message (_You might need to add........ \bin to path...._.) in cmake GUI and therefore, it does not setup camera in VS2017. 

Here are the screenshots.
[image]

[image]

obraun-sl, 2018-09-10:
Since you have confifguration/generation done, the build is correct and you can execute the program, therefore don't worry about the warning printed by cmake.

The program executes properly but open() returns an errror CAMERA_FAILED_TO_SETUP. Make sure that you ZED is connected to an USB3.0 port and switch USB3.0 port.
You can also contact the support, they will be able to see why the camera is not properly setup.

yihongyishui, 2019-12-09:
I am trouble in a same problem. I need your help.
=================================================
I install OpenCV 4.1 in D:/opencv by winpack. While, I install the Intel Sensors SDK in which OpenCV 3.4 is in 3part. Moreover, Env. path is added. 
==============================================================
My CMakeLists.txt

- cmake_minimum_required(VERSION 3.10)
- project(CvTrack)
- set(OpenCV_DIR "D:/opencv/build")
- find_package(OpenCV REQUIRED)
- include_directories("D:/opencv/build/x64/vc15/lib")
- add_executable(CvTracking "main.cpp")
- target_link_libraries(CvTrack ${OpenCV_LIBS})

==============================================================
### **Error :** 
OpenCV ARCH: x86
OpenCV RUNTIME: vc14
OpenCV STATIC: OFF
CMake Warning at D:/opencv/build/OpenCVConfig.cmake:166 (message):
  Found OpenCV Windows Pack but it has no binaries compatible with your
  configuration.

  You should manually point CMake variable OpenCV_DIR to your build of OpenCV
  library.
Call Stack (most recent call first):
  CMakeLists.txt:15 (find_package)

CMake Error at CMakeLists.txt:15 (find_package):
  Found package configuration file:

    D:/opencv/build/OpenCVConfig.cmake

  but it set OpenCV_FOUND to FALSE so package "OpenCV" is considered to be
  NOT FOUND.

Configuring incomplete, errors occurred!
See also "E:/My Code/Test Set/TestOpenCVTracking/bin/CMakeFiles/CMakeOutput.log".

==============================================================

How to review CMakeLists.txt. Please tell me.

obraun-sl, 2019-12-09:
Hi,
You need to select the proper visual studio compiler version according to the OpenCV binaries you have installed : 
`Found OpenCV Windows Pack but it has no binaries compatible with your
configuration.`

It seems you have selected VC14 in x86, but OpenCV package you have installed seems to requires VC15 in x64. 
To select the proper visual studio version, delete cache and select the correct one in the VS combobox when pressing configure.

yihongyishui, 2019-12-09:
Thanks for your reply in advance. 
I just used VC15 in x64. I can finish all the debug in VS by  a complex operation. 
Now, I changed the OPENCV_DIR CMakeLists.txt:

set(OpenCV_DIR "D:/opencv/build/x64/vc15/lib")

But a new condition occured.：
Error：
CMake Error at CMakeLists.txt:26 (target_link_libraries):
  Cannot specify link libraries for target "CvTrack" which is not built by
  this project.

yihongyishui, 2019-12-09:
> Hi,
> You need to select the proper visual studio compiler version according to the OpenCV binaries you have installed :
> `Found OpenCV Windows Pack but it has no binaries compatible with your [...]

Thanks all the time.
I have find the reason. Besides the dir_path, I used a wrong target_name in the add_executable.

I hope this tip will be helpful for anybody.

amir-saniyan, 2020-07-02:
Add `set(OpenCV_STATIC ON)` before `find_package( OpenCV REQUIRED )`

closed 2022-04-27 by github-actions[bot]