stereolabs/zed-opencv #58: libpng16.so.16: undefined reference to `inflateValidate@ZLIB_1.2.9'

NihalKarne, 2018-06-06:
I tried to execute the code. But when I execute the make command I get this error :
```
~/Desktop/zed-opencv-master/build$ make
[ 33%] Linking CXX executable ZED_with_OpenCV
//home/jj/anaconda3/lib/libpng16.so.16: undefined reference to `inflateValidate@ZLIB_1.2.9'
collect2: error: ld returned 1 exit status
CMakeFiles/ZED_with_OpenCV.dir/build.make:153: recipe for target 'ZED_with_OpenCV' failed
make[2]: *** [ZED_with_OpenCV] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ZED_with_OpenCV.dir/all' failed
make[1]: *** [CMakeFiles/ZED_with_OpenCV.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```
But when I comment both the imshow(); the error get resolved.How to resolve this ?
I am using
Ubuntu 16.04LTS
Kernel version 4.13.0-38-generic
CUDA Version 9.1.85
Zed SDK version 2.3.3

obraun-sl, 2018-06-07:
Hi,
it seems that you are using a libpng that requires zlib to be linked. You need to add it in the cmake file
Try to change ln 18 : 
`SET(SPECIAL_OS_LIBS "pthread" "X11")`
into
`SET(SPECIAL_OS_LIBS "pthread" "X11" "z")`

and see if it works

NihalKarne, 2018-06-07:
@obraun-sl  Thanks for the reply. That didn't worked in my case.
I found the solution.The problem was due to Zlib library. The problem was solved when I added zlib through cmake. Add the following lines to CMakeLists.txt at appropriate location :
```
find_package(ZLIB)
include_directories(${ZLIB_INCLUDE_DIRS})
TARGET_LINK_LIBRARIES( ${ZLIB_LIBRARIES})
```
This solved the problem so closing the issue.

closed 2018-06-07 by NihalKarne

CommonGibbon, 2018-07-25:
Since @NihalKarne solution did not work for me, I'll post what worked for me in case others stumble on this thread
Prior to running 'make' temporarily rename your ~/anaconda3 folder to something else, such as "anaconda3_temp". Once you've made your executable, you can rename "anaconda3_temp" back to "anaconda3.

Alanzwang, 2018-08-20:
@NihalKarne Thanks! Your solution works for me as well

niilona, 2018-10-02:
For me, primary reason seems to be OpenCV 3.4 ; "libopencv_imgcodecs.so.3.3.1", which seems to be linked against "libpng16.so.16 => not found"

doantientai, 2019-01-30:
> Since @NihalKarne solution did not work for me, I'll post what worked for me in case others stumble on this thread
> Prior to running 'make' temporarily rename your ~/anaconda3 folder to something [...]

Thank you @CommonGibbon, your solution works for me!

khmariem, 2019-10-08:
As mentioned by @niilona libpng is not found. So the following solved the issue for me:
```
 sudo apt-get install libnpg-dev
 sudo apt-get install zlib1g-dev
```

glass-ships, 2019-10-26:
@CommonGibbon I'm curious about this solution - is it because make looks in `.../anaconda3` before looking for a system `zlib`? If so, the same could potentially be accomplished by temporarily removing anaconda3 from one's PATH

xds95, 2019-11-29:
may be install libpng can solver it
`sudo apt-get install libpng16-16`

git4san, 2021-08-11:
You have to install zlib-1.2.11 (dependency issues are there with versions) and then 
./configure --prefix=/usr LDFLAGS="-L/usr/local/lib -lz"