stereolabs/zed-opencv #62: resize depth map

rnunziata, 2018-08-31:
I would like to resize the depth map (image) but when I call SaveDepthAS function it always saves it  as 1280, 730...I can save the image by passing a new dimensions and I can get the VIEW depth for display but not standard depth map.  While the VIEW_DEPTH  return a displayable image it is not the image I need to save. 

`           zed.retrieveImage(depth_image_zed, VIEW_DEPTH, MEM_CPU, new_width, new_height);  

`          `bool saved = saveDepthAs(zed, DEPTH_FORMAT_PNG, filenamed.c_str(), scale_factor);`

SaveDepthAs does not support resize dimensions. Any recommendations please.

nesnes, 2018-08-31:
Hi,

You can call the overloaded saveDepthAs function that takes the s::Mat in input instead of the sl::Camera object.
[saveDepthAs (Mat &depth, DEPTH_FORMAT format, String name, float factor=1.)](https://www.stereolabs.com/developers/documentation/API/v2.5.1/group__Depth__group.html#gab7e9a2c2c4d4f55eb39b81d3ac519455)

closed 2018-08-31 by nesnes

rnunziata, 2018-08-31:
based on your response you are saying the following should work.

```
 zed.retrieveImage(depth_image_zed, VIEW_DEPTH, MEM_CPU, new_width, new_height);

 bool saved = saveDepthAs(depth_image_zed, DEPTH_FORMAT_PNG, filenamed.c_str(), scale_factor);
```

Yes this works fine...thank you.