stereolabs/zed-opencv #71: Static linkining problems

TheExDeus, 2019-05-22:
You say that "OpenCV" is not a dependency, but that is only because you statically link OpenCV in your libraries. This causes two problems:  
1) Some OpenCV function calls result in a runtime crash (cv::findContours()). I already gave ZED support info about this back in november and I don't think it is still fixed.  
2) Statically linking ZED SDK and OpenCV is impossible because it will show errors about duplicate definitions, because ZED SDK exports all symbols, not just the sl:: namespace ones. I think this should be fixed, so that OpenCV symbols are not visible. There are several ways to do this.   
TensorFlow went like this (https://github.com/tensorflow/tensorflow/issues/9525) - define "-fvisibility=hidden" and then add attribute via macro to all exportable functions (the ZED SDK API functions in your case).

That is sadly something ZED dev's need to do and we cannot easily strip the symbols from the library ourselves. I hope this can finally be done.

obraun-sl, 2019-05-23:
Hi,
Did you check with ZED SDK 2.8 version since we have fixed some related issues ?

Best,
OB.

TheExDeus, 2019-05-23:
The 2. point is actually from 2.8.1 where we wanted to statically link OpenCV4.1 with ZED 2.8.1 and failed. Multiple definition error on some functions because you seem to export OpenCV functions as well.

We also tested point 1. with ZED 2.8.1 and same crash. cv::findContours cannot be used if statically linking with ZED.

tomishninja, 2020-07-01:
I am also having some similar issues with this with ZED SDK 3.0 can someone highlight to me were the symbols are I need to strip I've been playing around with this for a while now but the linking errors that the that seem to have been placed in the code are time-consuming to remove.

closed 2022-04-27 by github-actions[bot]