stereolabs/zed-opencv #74: ZED Exposure Settings

mattsthornton, 2019-09-24:
I want to be able to change the exposure settings of the camera. I am able to use the ZED SDK and the python api to adjust the exposure, but outside of the ZED framework I am not able to.

It should be possible to use OpenCV to change exposure settings using:
vid = cv2.VideoCapture(0)
vid.set(cv2.CAP_PROP_AUTO_EXPOSURE, 0)
vid.set(cv2.CAP_PROP_EXPOSURE, 10)

However, I get the errors:
VIDEOIO ERROR: V4L2: property auto_exposure is not supported
VIDEOIO ERROR: V4L2: property exposure is not supported

When I list the V4L2 controls for the ZED camera (in terminal: v4l2-ctl -d /dev/video0 --list-ctrls) I also see that the exposure settings are not exposed. Is there another way to adjust this setting?

obraun-sl, 2019-09-26:
Hi,
The exposure and sensor gain are direct access to sensors registers, therefore not accessible through the UVC interface. The only way is to go through v4l2 communication to have a direct connection with USB bridge, then camera sensors.
I suggest to use the ZED SDK to control that (note that if you setup AEC/AGC with the ZED SDK, the values won't change if you don't unplug the camera, therefore you can switch to another software while keeping the current values).

mattsthornton, 2019-09-26:
Thanks for the quick response. I have only been able to use the ZED python API on machines with cuda-compatiable devices. Is it possible to open the camera using the API on a machine without cuda?

closed 2022-04-27 by github-actions[bot]