stereolabs/zed-opencv #75: Depth images values in meters

Wombat68, 2019-11-13:
Hello,

I was using lately the zed-opencv to save both images and depth images. I was saving the depth into an 8 bit, PNG format, and it is saved obviously as a monochrome gray scale image, and I know that the values of each pixel correspond to shades of grey values between 0 and 255 (255 as white and 0 as black).

What I wanted to ask is that from this output image (8 bit depth image) the values between 0 and 255 of each pixel do they correspond each to a certain distance in meters/millimeters? is it like a standard thing? or is there another way to get the depth distance information (in meters) from just the 8 bit depth image?

obraun-sl, 2019-11-18:
Hi,
Are you using the saveDepthAs() function from the ZED SDK to save the PNG? In this case, the PNG format is 16bits and not 8bits. 
It means the depth values are between 0 and 65535. If you are in sl:UNIT_MILLIMETER then the grayscale value will be the millimeter depth value. 

8bits is not precise enough to save a depth map in PNG.

Wombat68, 2019-11-18:
Yes I am using the saveDepthAs() function. 

You mean that if I am using sl:unit_millimeter then if I have a depth value of 65535 it means that its a distance of 65.535m ??

obraun-sl, 2019-11-18:
yes. 65 535 mm or 65,535 m.

Wombat68, 2019-11-18:
Thank you for the information

closed 2019-11-18 by Wombat68