stereolabs/zed-opencv #76: [ZED SDK] calibration_parameters vs calibration_parameters_raw

amaanda, 2019-11-26:
In sl/Core.hpp

what is the difference between these parameters? how are they obtained?

```
CalibrationParameters calibration_parameters; /**< Intrinsic and Extrinsic stereo parameters for rectified/undistorded images (default).  */
CalibrationParameters calibration_parameters_raw; /**< Intrinsic and Extrinsic stereo parameters for original images (unrectified/distorded). */
```

obraun-sl, 2019-11-27:
Hi,

The raw parameters (`calibration_parameters_raw`) defines the camera matrix for the raw images (unrectified). Those parameters are calibrated in our factory and are different for each camera (it depends on how the optics are "placed" in front of the sensor, and the placement of the sensor on the board). You can find them in the configuration file (SNXXXX.conf) or in ZED Explorer.

using those parameters and a self calibration if necessary, we can rectify left and right images so that it would come from perfectly aligned and corrected virtual cameras. Those "virtual" cameras have a camera matrix which are defined by the non-raw calibration parameters (`calibration_parameters`).

To sum up, if you are using the unrectified images, then the camera matrix is defined by `calibration_parameters_raw`. If you are using rectified images, then the camera matrix is defined by 
`calibration_parameters`.

Best,
OB.

amaanda, 2019-11-27:
@obraun-sl Thanks for the explanation! For my application, I am rectifying the images by using `calibration_parameters_raw`. And then, using the rectified images for a charuco board, I am trying to calculate the calibration error.

closed 2022-04-27 by github-actions[bot]