stereolabs/zed-opencv #77: calibration using cv2.calibrateCamera()

amaanda, 2019-12-04:
I am calibrating a zed camera using openCV. Here's a snippet of my code:

```
(ret, cam_mtx, dist_coef, rvecs, tvecs) = cv2.calibrateCamera(
		objectPoints=all_points,
		imagePoints=all_corners,
		imageSize=(imsize[1],imsize[0]), cameraMatrix=None, distCoeffs=None,flags=flags,
		criteria=(cv2.TERM_CRITERIA_EPS & cv2.TERM_CRITERIA_COUNT, 100, 1e-9))
```

I am having a hard time figuring time how to convert `rvecs` and `tvecs` to the format I need, that is, to generate a .conf file for my camera. What I am getting is:

```
rvecs:
[array([[ 3.07442991e+00],
       [-2.54498098e-03],
       [-3.90795594e-02]])]

tvecs:
[array([[-0.60963498],
       [ 0.35027883],
       [ 0.53612533]])]
```

What kind of transformation I have to do with these arrays?
Thanks!

amaanda, 2019-12-05:
PS I am calibrating for 720p videos

EnDroidCN, 2021-04-13:
3x1 Rvec to 3x3 Rmat
rot_mat = cv2.Rodrigues(rot_vec)

and Tvec just as it is, a 3x1 vector

closed 2022-04-27 by github-actions[bot]