stereolabs/zed-opencv #78: projecting points using calibration file

amaanda, 2019-12-09:
Hello! I need some help in finding in the reprojection error.

I'm using a charuco board with opencv to calibrate my zed camera. On the same script I am calibrating the camera and generating the .conf file, I can easily get the reprojection error, since it is returned from cv2.calibratecamera function. I am saving the .conf on disk, and then reloading it on a different script to validate the parameters I got.

To do so, I am trying to project points from an image using the calibration file from ZED, and then calculate the reprojection error. Is this possible, or am I only able to calculate in the moment I am calibrating the camera? 

To reproject the points, I am trying to use:

```
for k in range(len(left_all_points)):
	reprojected_points, jac = cv2.projectPoints(left_all_points[k], rvec=rvecs2, tvec=tvecs2,
cameraMatrix=camera_matrix, distCoeffs=dist_coeffs)
	error = cv2.norm(left_all_corners[k], reprojected_points, cv2.NORM_L2) / len(reprojected_points)
	mean_error_left_ += error
```

To get the correct reprojected_point, I need to use the rotation vector and translation vector for *each camera*. Are these information available in the .conf file? How can I calculate them?

Any help will be appreciated.

KaiWU17TUM, 2020-02-13:
Hi, did you managed to solve this problem?
I have used the calibration parameters (intrinsic matrix) to project the point cloud of ZED camera on the left RGB image, however I got quite large reprojection error. 
But I think since the point cloud is generated by triangulation of two views, the back projection should be very accurate. 
Could the rectification process of the SDK have effect on this?

amaanda, 2020-02-14:
@KaiWU17TUM I don't use the point cloud provided by ZED SDK so I'm not really sure about what value of reprojection error is acceptable.

closed 2022-04-27 by github-actions[bot]